SpenceKonde/megaTinyCore #228: Guidance required on getting RS485 to work on ATtiny3216

MichielVanwelsenaere, 2020-09-25:
First of all, thank you for your great work @SpenceKonde!

For the past months I've been working on a custom PCB board for the ATtiny3216 to act as a sensorhub which can be connected to leveraging Modbus RTU over RS485.

Repo can be found [here](https://github.com/MichielVanwelsenaere/HomeAutomation.ModbusRtuSensorHub).

I'm very close to delivering a V1.0 version of the project but am currently struggling to get any Modbus RTU arduino library to work.

According to the datasheet of the ATtiny3216, RS485 is natively supported and there's no need for an external transiever chip. 

As I'm having a hard time finding it in the repo over here could you elaborate on the following:
- does 'megaTinyCore' enable RS485 by default on the TX/RX pins or is additional setup (like setting a register) required?
- as the ATtiny3216 supports RS485 natively which is not the case for regular arduino boards (at least the ones I know of) what are the chances that existing arduino library will do the trick?

Thank you for your input, much appreciated!

SpenceKonde, 2020-09-25:
The ATtiny 0/1/2-series (as well as megaAVR 0-series and new AVR Dx series) does *not* support RS485 without a transceiver. What it does do is provide native support for controlling the direction/tx-enable pin of the transciever on the XDIR pin - you would just connect that pin to the corresponding pin of your RS485 transciever.

It is not enabled by the builtin HardwareSerial implementation - this would be a disaster, because it would take over control of that pin, which hardly anyone wants to use!
After calling the begin() method of the serial port in question, you just need to enable it by setting the appropriate bit:
Ie, assuming USART0 (Serial), something like

```c 
Serial.begin(115200);
USART0.CTRLA|=USART_RS485_INT_gc;
```

Note that you must have also set the XDIR pin to output first. Future versions of megaTinyCore will provide defines of the form PIN_HWSERIAL0_XDIR and PIN_HWSERIAL0_XDIR_PINSWAP_1 - but this is not yet in any version of the core; I'd suggest something like this at top of sketch, so you can just do `pinMode(PIN_HWSERIAL0_XDIR,OUTPUT);` in setup.

```c
#ifndef PIN_HWSERIAL0_XDIR
#define PIN_HWSERIAL0_XDIR PIN_PA4
#endif
```

SpenceKonde, 2020-09-28:
No further responses from @MichielVanwelsenaere - closing issue as this does not represent a defect in the core; I will still see and respond to closed issues.

closed 2020-09-28 by SpenceKonde

MichielVanwelsenaere, 2020-09-28:
Thank you for your feedback @SpenceKonde! Haven't gotten the chance to try it out yet. Will revert when done so.

MichielVanwelsenaere, 2020-09-30:
I progressed on this a bit more by wiring a TTL to RS485 convertor in between and using the sketch below:

```C
#include <ModbusRtu.h>

#ifndef PIN_HWSERIAL0_XDIR 
#define PIN_HWSERIAL0_XDIR PIN_PA4
#endif

#define SLAVEID 1 // ModBus Slave adres.
#define BAUDRATE 9600 // ModBus Slave baudrate.
 
// Modbus register data array (start register = 0)
uint16_t au16data[16] = {3, 14, 9265, 4, 2, 7182, 28182, 8, 0, 0, 0, 0, 0, 0, 1, 28}; // 16 registers
 
/**
 *  Modbus object declaration
 *  u8id : node id = 0 for master, = 1..247 for slave
 *  u8serno : serial port (use 0 for Serial)
 *  u8txenpin : 0 for RS-232 and USB-FTDI or any pin number > 1 for RS-485
 */
Modbus slave(SLAVEID, Serial, PIN_HWSERIAL0_XDIR); // Stel slave ID en protocol in
 
void setup() {
  pinMode(PIN_HWSERIAL0_XDIR,OUTPUT);
  Serial.begin(BAUDRATE, SERIAL_8N1); // Begin slave met baudrate en poortinstellingen
  USART0.CTRLA|=USART_RS485_INT_gc;
  slave.start();
}
 
void loop() {
  slave.poll(au16data, 16);
}
```

After uploading it I use QModbusMaster to launch requests to the board. I've tried all the possible read requests (1 to 4) but keep getting a timeout.
I double checked the wiring and also verified with a LED on the data wires if the requests are send to the board (they are).
Yet, no luck, all I get is a persistent timeout (already tried to increase the timeout value).

After that I checked the Modbus library that I'm using (can be found [here](https://github.com/smarmengol/Modbus-Master-Slave-for-Arduino/blob/master/ModbusRtu.h)) to see if anything is coded that makes it unusable with ATtiny3216 but couldn't spot anything.

Would you have an idea on how to troubleshoot this further?

Thanks for your input!

MichielVanwelsenaere, 2020-10-13:
for anyone attempting something similar: changing the XDIR pin to PIN_PC2 got things working for me.

SpenceKonde, 2020-10-13:
What the helllll? Can you share code that demonstrates the behavior and a quick sketch of the wiring? There is no way that using the builtin USART XDIR functionality should *ever* work on PC2!

SpenceKonde, 2020-10-13:
Maybe I'm misunderstanding what you're trying to do...?

MichielVanwelsenaere, 2020-10-14:
[image]

This is the wire diagram. for all clarity; 
- naturally both of the boards need to be powered
- the PC2 pin is connected to both DE & RE