SpenceKonde/megaTinyCore #230: fix the do_spm vector in optiboot...

WestfW, 2020-09-27:
I think I have this working.
https://github.com/Optiboot/optiboot/commit/7975070bdc1fb42c46713070a01c05971c37ec7d
Neat test program, too.

(I could've saved some time if I'd looked at the stuff you'd already done.  Sigh.)

SpenceKonde, 2020-09-28:
Thaaaaanks! So glad to see we finally have this working! - marking this for 2.0.6; the library wrapper around it will be 2.1.0 though.

WestfW, 2020-09-28:
Now updated: https://github.com/Optiboot/optiboot/tree/master/optiboot/examples/test_dospm
The optiboot.h in here now defines do_nvmctrl() for the mega0 chips.
Better, it defines some "higher level" functions like:

```
 | * optiboot_page_erase - to erase a FLASH page                            |
 | * optiboot_page_fill - to put words into temporary buffer                |
 | * optiboot_page_write - to write contents of temporary buffer into FLASH |
 | * optiboot_readPage - higher level function to read a flash page and     |
 |                         store it in an array                             |
 | * optiboot_writePage - higher level function to write content to         |
 |                         a flash page             
```
that I believe will now work on both traditional AVRs and mega0/xTiny with identical code.
the test_dospm.ino sketch seems to work on both Uno and ATtiny3217, so that's a pretty good sign.

note that these high-level functions are use 0 as the start of PROGMEM, while the do_nvmctrl() lower-level functions need to be passed the aliased addresses.

WestfW, 2020-09-28:
you don't seem to have optiboot_x source in your core; just .hex files.
Are those built with the makefiles and targets that I made, or did you build them differently?
I am nervous about doing a PR for .hex files!

SpenceKonde, 2020-09-28:
Wait, wtf, the source is supposed to be in the core....

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

WestfW, 2020-09-28:
Hmm.  It's there on github.  Somehow the source disappeared from my local clone :-(

SpenceKonde, 2020-10-02:
Fixed by #233

closed 2020-10-02 by SpenceKonde