SpenceKonde/megaTinyCore #231: Full implementation of neopixel library

intakenick, 2020-09-27:
Are there any plans to implement the remaining missing functions in the neopixel library like ColorHSV or fill()?

SpenceKonde, 2020-09-27:
"missing functions"? I think you mean "new functions" (er, technically they're "methods", I believe)  - they were added to the adafruit library after tinyNeoPixel was forked from it, and I was entirely unaware of their existence.

The methods that operate on the pixels themselves are identical, adding these will be trivial.

intakenick, 2020-09-27:
thanks for the response and good callout, "new methods" is what I should have said! I figured as much. I'll take a crack at adding them in and will circle back if I'm successful. Thanks for all of your hard work!

SpenceKonde, 2020-09-27:
Hmm, they've added more than just those too....

SpenceKonde, 2020-10-02:
In github version now. (along with untested support for overclocked parts, pulled from DxCore)

closed 2020-10-02 by SpenceKonde