SpenceKonde/megaTinyCore #237: RTC millis rate conversion?

thomlim, 2020-10-04:
Hi! In wiring.c in the millis() function, there is this line to convert the RTC timer rate:

    //now correct for there being 1000ms to the second instead of 1024
    m = m - (m >> 5) - (m >> 6);

This looks like its converting it down to 976, not 1000.  Should it actually be:

    m = m - (m >> 6) - (m >> 7);

or am I misunderstanding what's happening here?

SpenceKonde, 2020-10-04:
RTC millis source is being removed in 3.0.0 (the next version). It was a bad way to implement millis keeping time during sleep, and I have regretted it ever since. Most of the plumbing is already in for what I'm going to do with this - there's going to be a library for sleeping with the RTC keeping time, and setting millis up when it wakes. All of the RTC millis code currently in the core is going out.

SpenceKonde, 2020-10-04:
And as far as I'm aware, that was the only reason people would use the RTC for millis

closed 2020-10-04 by SpenceKonde

SpenceKonde, 2020-10-04:
RTC option has gotten a reprieve as the major changes involved in this update were canceled for other reasons. Note that this option IS still going away in the future, pending availablity of that library. You are correct about that code, it is fixed now.

thomlim, 2020-10-05:
Ok thanks.
Yeah, looking forward to the sleep library, thanks!