SpenceKonde/megaTinyCore #238: MEMS as External clock

Yardie-, 2020-10-05:
This is more a couple of questions rather than an issue.
I am thinking of using a SiT1552 32.768 kHz MEMS as an external clock source.
Most specifically to implement a 3216 as an rtc with a screen.
I have several reasons for this that I won't bore you with but am happy to explain if required.

Q1. should I connect to PA3 EXTCLK or PB3 TOSC1 the datasheet seems to suggest both are possible 
I see your readme suggests TOSC1 and 2 for an oscillator however these only have one output and I am guessing that they work the same as a normal 32.768 output of an rtc. ( I could be barking up the completely wrong tree here ) 

Q2. Is this a bad idea ie what are the potential problems.

Thanks !
:) Greg

SpenceKonde, 2020-10-06:
Q1: TOSC1 - EXTCLK is for running the chip at 32.768 kHz, this is not supported by megaTinyCore (or anything implementing the Arduino API or an approximation thereof - the user experience would be utterly miserable - digitalWrite() would take several milliseconds to execute, for example)
Q2: When using an external oscillator, rather than an external crystal for the 32.768kHz RTC clock, you need to set the SEL bit to 1 in XOSC32KCTRLA (this isn't supported with the RTC as millis source, which has numerous problems and needs to be rearchitected)

Yardie-, 2020-10-07:
Thanks heaps Spence that clarifies it a lot.
I read over the datasheet a few times before I asked.

So at the moment the most practical solution is to stick with a crystal so that millis still works.
The fabrication at jlcpcb.com is a little problematic at the moment as their footprint is broken.
So the easier way seems the most correct for now. 
( often the easy way is the best but I say wheres the fun in that? ) 

one thing I couldn't see either was 
If I used an external oscillator would that free up TOSC2 for GPIO ?

Maybe this is something to play with with the next gen chips. 

Anyway thanks again Spence you are a legend.

:) Greg

SpenceKonde, 2020-10-09:
Yes it would free up the TOSC2 pin. 

I'm throwing in support for external oscillator as RTC clock source into 2.1.0

Yardie-, 2020-10-09:
Excellent. Maybe it will translate directly into the next gen chips as well. Thanks Spence.

SpenceKonde, 2020-10-10:
Implemented. Yes, it will not need any change for 2-series tinyAVR. Note that I am holding off on implementing it on DxCore until I make some changes I want to make. (among other things, make it accurate to only milliseconds in sleep, but while awake, it could be accurate to 1/32768'ths of a second, meaning micros() could be made available with a resolution of 30.5 uS - part of my modernSleep library.

closed 2020-10-10 by SpenceKonde