SpenceKonde/megaTinyCore #239: Intermittent Stuck bits on UART for 2.0.2 , and 2.0.5 - NOT on 2.0.1 Attiny412

ldalep, 2020-10-09:
I am trying to make a poor mans WS2811 using the Attiny412.  This will then be driven by the neoPIXEL library with the option that allows 115200 serial out.  The code seem to work just fine using 2.0.1  but at 2.0.2, 2.0.5 it gives me intermittent high bits.  This happens at 20mhz, 10mhz,  and 16mhz.....  

[image]

//#include <SoftwareSerial.h>

//const byte rxPin = PIN_PA6;
//const byte txPin = PIN_PA5;

const byte P[3] = {PIN_PA3, PIN_PA1, PIN_PA2};
byte MyByte[3] = {0, 0, 0};
byte Tmpbyte = 0;

unsigned long  t = micros();
byte ii = 0;

//SoftwareSerial InfoSerial (rxPin, txPin);

void setup() {

  pinMode(P[0], OUTPUT);
  pinMode(P[1], OUTPUT);
  pinMode(P[2], OUTPUT);

  Serial.begin(115200);
  t = micros();
}

// the loop routine runs over and over again forever:
void loop() {

  while ((micros() - t)  < 1100) {
   
    if (Serial.available() > 0) {   
      if (ii < 3 ) {
        MyByte[ii] = Serial.read();
        Serial.print( MyByte[ii] ); Serial.flush(); 
      }
      else
      {
        Tmpbyte =Serial.read();
        Serial.write(Tmpbyte); Serial.flush();
      }
      ii++;
      t = micros(); 
    }

  }
  
  analogWrite(P[0],   MyByte[0]);
  analogWrite(P[1],   MyByte[1]);
  analogWrite(P[2],   MyByte[2]);
  ii = 0;
  t = micros();
}

SpenceKonde, 2020-10-10:
Where is the data coming from?

SpenceKonde, 2020-10-10:
As I see *two* weird things there - the doubled 1, and the additional character....

ldalep, 2020-10-10:
The Data is coming from Cool Term - Simple string of hex data  00 00 01 01 02 03 04 05 06  .... 18  via a mac out a serial port into the RX of the ATTiny412 ..... The first 3 bytes are stripped off and echoed back as text from Attiny's TX port and     the remaining incoming data is sent back as simple hex (write vs print) -  in this example Cool Term  receiving data has 31 32 instead of 01 02 (circled in red).  This type of error can happen anywhere in the data stream.  When I run the same pattern/sketch using 2.0.1 I don't have any errors.    I think I paid  .43/part and the nice thing about these are I can put them to sleep and I don't have to invert the RGB outputs to the LED drivers.  I haven't tried the same correlations at lower bauds but I am stuck at 115.2K because that is what the NeoPixel library uses for thier high powered leds.  (not ws2811)

SpenceKonde, 2020-10-10:
Hmm.... What serial adapter is this?

ldalep, 2020-10-10:
I have tried two different ones with the same results - "DSD TECH USB to TTL Serial Adapter with CP2102 Chip Compatible with Windows 7,8,10,Linux,Mac OS X"  and  "DSD TECH SH-U09C2 USB to TTL Adapter Built-in FTDI FT232RL IC"

ldalep, 2020-10-10:
A little note for what it is worth..... I use the first 3 bytes as colors for module then pass the rest on.  If you notice that I use the Serial.print( MyByte[ii] ); Serial.flush(); this is only for debug.  Again this works fine for version 2.0.1 no errors.  For version 2.0.2 and 2.0.5 I get errors.  If I change this to Serial.write ( MyByte[ii] );  I see no errors in the various 3 version

SpenceKonde, 2020-10-10:
Well that's a nice clue.... 

Sadly this missed the window for 2.1.0 - this is 2.1.x or 2.2.0

ldalep, 2020-10-11:
OK.... I have a confession - it isn't quite true that 2.0.2, 2.0.5 completely works when switching to a simple Serial.write.  Turns out that when I added the <tinyNeoPixel_Static.h> and modified the show to produce serial out I had random flickers when going through smooth gradients of color.  Well going back to 2.0.1 FIXED that too.  sigh......

ldalep, 2020-10-11:
Also tried 2.1.0 - I couldn't get the following to work - nothing ??? what am I missing

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello");
}

void loop() {
   
    if (Serial.available() > 0 ) {
        Serial.print(Serial.read());Serial.flush(); 
       
    }
}

SpenceKonde, 2020-10-13:
Serial was broken on 8-pin parts in 2.1.0, 2.1.1, and 2.1.2, just released 2.1.3 to fix that and burn bootloader for 8-pin parts. 

2.1 was not my most graceful release...

ldalep, 2020-10-13:
I’m amazed at what you do ..... no apologies needed!!!

SpenceKonde, 2020-10-17:
If you have a moment can you check USART0.BAUD on the two versions? Did that change? (asking you to do it, because it could vary based on the oscillator corrections which are chip-specific). 

If they are different, try putting 
`USART0.BAUD=(old USART0.BAUD)` after Serial.begin and see if that fixes it. 

Though that wouldn't really make sense if it's failing in 2.0.2, not 2.0.3 (the lost release) /2.0.4 because the only changes that I can imagine breaking it were in 2.0.3/2.0.4.... I tweaked the USART baud rate calculations at that point.... But even that doesn't seem to match what you're seeing....

ldalep, 2020-10-17:
I only tried 2.0.1, 2.0.2, 2.0.5, 2.1.3  -  I muddied the water by saying that the problem appeared to be fixed by changing from serial.print to serial.write seemed to have fixed the problem.   It helped but when I started to stream data from the tiny804 the only version that was clean was 2.0.1  - Does that help?   basically what could be different between 2.0.1 and 2.0.2

ldalep, 2020-10-17:
should say streamed data from the tiny804 to the tiny412   changes on only the tiny412 where made

SpenceKonde, 2020-10-17:
Hmm, wait, so it's 2.0.1 vs 2.0.2 that's the difference?! 

That's wacky.... that's when I was lost in the twilight zone of timer stuff...

ldalep, 2020-10-17:
Yes my app works in 2.0.1  but versions after don't

SpenceKonde, 2020-10-17:
Weird...

ldalep, 2020-10-17:
Yup I am stuck in 2.0.1 outer limit zone

SpenceKonde, 2020-10-17:
I can't look at this until after DxCore 1.2.0 is ready

ldalep, 2020-10-17:
OK....even tried a different 412 chip

SpenceKonde, 2020-10-17:
Are any of the USART0 registers any different? If they are, that's a huge clue.

But like, I didnt' change much between 2.0.1 and 2.0.2...

ldalep, 2020-10-17:
Ok I hacked this  I don't know it is of any help - I put these prints at the end of setup  on 2.0.1 - compiled and results are at the side  (No Errors)

  Serial.println (USART0.RXDATAL, HEX);  -> 0
  Serial.println (USART0.RXDATAH, HEX);  -> 0 
  Serial.println (USART0.TXDATAL, HEX); -> 30
  Serial.println (USART0.TXDATAH, HEX); ->0
  Serial.println (USART0.STATUS, HEX);->0
  Serial.println (USART0.CTRLA, HEX);  ->A0
  Serial.println (USART0.CTRLB, HEX);-> C0
  Serial.println (USART0.CTRLC, HEX); -> 3
  Serial.println (USART0.BAUD, HEX); -> 2BF 
  Serial.println (USART0.reserved_0x0A, HEX);  ->   0
  Serial.println (USART0.DBGCTRL, HEX);  -> 0
  Serial.println (USART0.TXPLCTRL, HEX); -> 0
  Serial.println (USART0.RXPLCTRL, HEX); -> 0 
  Serial.println (USART0.reserved_0x0F, HEX);- >0

Did the same with 2.0.2 - verified there was errors and got the same as above.

As an aside I decided to run the same using baud from 115200 to 57600   and repeated the above.

Got the same results between 2.0.1 and 2.0.2 but both changed as probably expected  
 (USART0.TXDATAL, HEX); -> A
 (USART0.BAUD, HEX); -> 57F

Dale 👎

SpenceKonde, 2020-10-17:
Thanks.... so it's not related to the configuration of the USART! That is rather more alarming, frankly....

ldalep, 2020-10-17:
Anything I can try - let me know

SpenceKonde, 2020-10-17:
What's it doing with micros there? can you describe the purpose of that and the significance of that 1.1ms freqency? 

Which timer do you use for millis/micros? Does picking a different one change anything?

The next thing I would do is try to figure out exactly what part of it was glitching. I'd remove the analogWrite()'s and assuming the bug didn't run and hide at that point, use those pins to get a signal out, eg, if it reads in a 30 in the top section, drive one line high, if it reads in a 30 in the other section drive another high? And maybe the third if ii gets higher than it's expected to....? for start? You already have LEDs on the pins too!

ldalep, 2020-10-18:
OK... I have some info.  The micros is there because it syncs up when the PWMs are to be updated at the same time with subsequent modules.  There is a >1msec pause in the data flow into the RX port at that point all other modules should have process their first three bits and passed the others on to the next. 

For the next clue what I have always done was to watch the 3 color LED for any inconstancy with its change in gradient or intensity as the data would go through the color wheel.   2.0.1 perfect result where >= 2.0.2 would flicker and stumble.  I "assumed" that this was bad data being processed.  Sorry that is not the case.  I took traces of 10Kbytes of input and output data and after syncing  the offsets I found no inconsistency between data IN and Out.   The Data out was only corrupt when  I use the Serial.Print as noted in earlier thread.  

Maybe the best clue is  that when you change to TCB0 clock using rev 2.0.1, 2.0.2 you get compile errors - Not so when you go to 2.1.3  -  Using TCB0 instead of the default fixes the problem and gives me the results that I have in 2.0.1 using the default clock TCD0

SpenceKonde, 2020-10-27:
This sounds like something out of the twilight zone or something. 

Serial.print and Serial.write absolutely shouldn't be getting different results here, lol... 

Would you mind giving it a try in 2.1.4? Quite frankly, any pre-2.1.4 USART related things are almost inapplicable to 2.1.4 and later. Compile size is down by as much as 380b on some parts and with favorable conditions in the rest of the sketch, and a few things that I considered bloated or of high-risk design were reworked extensively; I totally redid how it deals with things being called in an ISR and how they interact with ISRs. I am convinced that the way they did it was asking for trouble; it would be quite gratifying that it turned out that in your use case, it was indeed finding trouble.

SpenceKonde, 2020-10-28:
2.1.4 is out. 

Based on the nature of the issue described, it is highly likely that the cause was a bug in UART.cpp, or at least that the code in that file was intimately involved in the cause of the problem. 

UART.cpp has seen major changes; it is not implausible that some of the changes I made may have fixed the problem you are seeing; some of the changes I made were made in part because I was concerned about the potential for strange interactions with other interrupts (though I still can't imagine how it could have done *THIS*). In any event, the magnitude of the changes is such that any observations made with the prior version that relate to serial in any way are no longer valid. So in order to determine what, if anything, should be done relating to this issue; if we are lucky, the only thing left to do is close the issue :-)

ldalep, 2020-10-28:
So I am following the interrupt camp - switching clocks seem to have totally fixed the issue ( pwm being intermittent) 2.1.3 - it will be a couple days before I can test 2.1.4 because I am trying “boot up” a left knee which I had replaced

ldalep, 2020-10-31:
So I tested 2.1.5 and TCD0 as then default timrt - Still has inconstant or non smooth transitions on the  PWMs  if I use the TCB0 which breaks tone the PWMs are smooth and responds as expected

SpenceKonde, 2020-11-01:
Does it fix the Serial.print() issue though? That's the one I was expecting it to fix, and which represents a very high priority issue if not... either it's receiving or sending characters it's not supposed to be, and either way, that's like... really bad... 

The PWM glitch didn't really even register with my because of the severity of the serial issue... Can you explain what it's actually doing? Is it a glitch lasting just one PWM cycle? If so I know what's causing it and how to fix...

ldalep, 2020-11-01:
See my comment 14 day’s ago - it appears that the serial has been fixed, at what point I am not sure.  The pwm glitch was probably there too.   As you suggested using different timers take the pwm issue from being broken to fixed .   My suspicion is that it is lasting one cycle but to verify that given the data stream will have to be later - however a change based upon your hunch I can visually determine

SpenceKonde, 2020-11-17:
Hmm... If PWM issue persists, let's break that off into a new issue; I did just make some subtle (from timekeeping perspective) changes to TCD, but I'm not sure they would have impacted parts where that's not used for PWM. I still have to say, I don't really understand that issue.

The single compare cycle problem I mentioned? The method to fix it doesnt work in split mode, which is used on all parts to get more compare (PWM) channels... though, one could argue that the 412, which uses it just to get one additional PWM channel maybe *shouldn't*, that's a job for a separate library... (oh - that's interesting... it would mean that init() needs to be split into a couple of weakly defined sub-functions for each part of the initialization process, otherwise the result would be hopelessly bloated, too bloated for the small flash parts... which it should be *anyway* now that I think about it... and I started that for starting, stopping, and initializing millis for the same essential reason to enable my magic sleep library which I, ah, never actually wrote... (though when I finally do write it, it'll work on all the post-2016 AVRs - there's nothing essentially different between them, especially since the core would keep control of how millis is implemented - same would be true about a TCA-magic library...). *definitely* don't wait on such a library, lol... If using TCB0 as millis clock works, that's probably the proper solution.

closed 2020-11-17 by SpenceKonde