SpenceKonde/megaTinyCore #240: Serial Output Behaves Differently, ATTiny416 vs. ATTiny816

craigfoo, 2020-10-09:
I'm working with both ATTiny416 and ATTiny816. I've been working with ATTiny816-MNR but it seems that it's availability is currently lacking. I'm able to fit my code on the ATTiny416 but the serial output is jumbled as if the baud rate is wrong. The two chips are pin for pin compatible, just different memory sizes.

I've tried changing the UART pins as well. 

```
//#define tinyTX    14   // PA1
//#define tinyRX    15    // PA2

#define tinyTX    7   // PB2
#define tinyRX    6    // PB3

void setup() {
  Serial.pins(tinyTX, tinyRX);
  Serial.begin(9600);
  Serial.println("Serial Setup");
 
}

void loop() {

  Serial.println("This is a test MFer!");
  delay(1000);
}
```

ATTiny416 Output
[image]

ATTiny816 Output
[image]

SpenceKonde, 2020-10-10:
Did you remember to "burn bootloader"? You need to do that to switch
between 20-mhz-derived clocks and 16-mhz-derived clocks. Note also that
automotive (-VAO) parts only support 16-muz-derived clocks.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

craigfoo, 2020-10-12:
Annnnd I didn't remember to burn bootloader. Did that and it worked. Thanks!

closed 2020-10-12 by craigfoo