SpenceKonde/megaTinyCore #241: Servo_megaTinyCore lib misses #endif

mrWheel, 2020-10-10:
The Servo_megaTinyCore misses an ```#endif``` at the end of the file..

SpenceKonde, 2020-10-10:
Thanks for reporting. I'll fix the #endif - fix for the underlying issue won't make it into 2.1.0, which is being released shortly.

I definitely *did* test it though.... I swear it used to work!
You're not by any chance using an "analog" servo? See #195

mrWheel, 2020-10-10:
No, not using an analog servo.

Tested it with #include <Servo.h> on an Arduino UNO. Servo’s work with that setup.

Regards,

Willem

gsm: 0653 - 499 551
www: http://www.aandewiel.nl/
---
If all men were brothers, would you let one marry your sister?

SpenceKonde, 2020-10-10:
Weird. Also, I can't find any missing #endif there???

mrWheel, 2020-10-10:
```
Arduino: 1.8.13 (Mac OS X), Board: "ATtiny3216/1616/1606/816/806/416/406, ATtiny3216, 20 MHz, 4.2V (20 MHZ or less), Disabled, Disabled, EEPROM retained, Enabled (default timer), Closer to 5v"
```
  
   
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/WillemA/Library/Arduino15/packages -hardware /Users/WillemA/Documents/ArduinoProjects/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/WillemA/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/WillemA/Documents/ArduinoProjects/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=20,bodvoltage=4v2,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -vid-pid=2341_0043 -ide-version=10813 -build-path /Users/WillemA/tmp/arduinoBuild -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /var/folders/lt/twpr82k512n1y9dwlck19hp40000gn/T/arduino_modified_sketch_510225/Sweep.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/WillemA/Library/Arduino15/packages -hardware /Users/WillemA/Documents/ArduinoProjects/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/WillemA/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/WillemA/Documents/ArduinoProjects/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=20,bodvoltage=4v2,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -vid-pid=2341_0043 -ide-version=10813 -build-path /Users/WillemA/tmp/arduinoBuild -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/WillemA/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /var/folders/lt/twpr82k512n1y9dwlck19hp40000gn/T/arduino_modified_sketch_510225/Sweep.ino
Using board 'atxy6' from platform in folder: /Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5
Using core 'arduino' from platform in folder: /Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5
Warning: Board standalone:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
Warning: Board tiny:avr:attiny85 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85
Warning: Board tiny:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
Detecting libraries used...
/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=20000000L -DARDUINO=10813 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -DUARTBAUD5V -I/Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/cores/arduino/api/deprecated -I/Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/cores/arduino -I/Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/variants/txy6 /Users/WillemA/tmp/arduinoBuild/sketch/Sweep.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Servo_megaTinyCore.h: [Servo_megaTinyCore@1.1.7]
ResolveLibrary(Servo_megaTinyCore.h)
  -> candidates: [Servo_megaTinyCore@1.1.7]
/Users/WillemA/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=20000000L -DARDUINO=10813 -DARDUINO_ [... cut ...]
  int readMicroseconds();            // returns current pulse width in microseconds for this servo (was read_us() in first release)
  bool attached();                   // return true if this servo is attached, otherwise false
private:
   uint8_t servoIndex;               // index into the channel data for this servo
   int8_t min;                       // minimum is this value times 4 added to MIN_PULSE_WIDTH
   int8_t max;                       // maximum is this value times 4 added to MAX_PULSE_WIDTH
};

#endif  // !defined(ARDUINO_ARCH_STM32F4)
#endif  // added by AaW ifndef Servo_h
```

SpenceKonde, 2020-10-10:
Aaaha! the errant ARDUINO_ARCH_STM32F4 is removed in github version, not sure when...

SpenceKonde, 2020-10-10:
You have confirmed that it's not a power issue right? Servos have a reputation for putting a lot of noise onto power rails (something like having blink or some serial prints going on at the same time, so you can see if it's resetting when it tries to move the servo?)

mrWheel, 2020-10-10:
Well,
I just noted that the servo’s (tiny little servo’s for a small tilt/pan construction) did not move at all with the example code “sweep” for the ATtiny3216. But now I see that my example’s are also not the same as those on github (for example” sweep includes Servo.h and I had to change it to “Servo_megaTinyCore” for it to compile).

I loaded and updated the megaTinyCore about Two weeks ago but I now think to remove it completely and then re-install it.

ccameron4k95, 2020-10-21:
I'm also seeing abnormal behavior with the Servo_megaTinyCore library v2.13 (updated in boardmanager), which I think is related to mrWheel's issue. Attached are images from an oscilloscope running the "sweep" example code on an ATtiny412. The output appears to be at a constant 50% duty cycle instead of a constant period (20 ms) with varying duty cycle (~1000-2000 us).

[image]
[image]

A Hitec HS-5065MG+ digital servo moves with that output, but it's definitely not working as intended.

SpenceKonde, 2020-10-27:
Yuck! That's nothing like how it looked when I wrote it!

ccameron4k95, 2020-10-27:
I think the problem is that TCB is running in periodic interrupt mode and resets to 0 every time it fires. I'm close to a functioning solution by declaring tcCounterValue globally and using it to accumulate the high ticks from each servo to determine the ticks remaining in the refresh period. The behavior is much closer to correct but I still have a small bug to work out where the frequency varies by around 20 Hz when changing pulse width.

SpenceKonde, 2020-11-15:
There are like a million problems here. 

The *real* question is how I *ever* managed to "test" it and get results that convinced me this steaming pile of shit was anything resembling working!

commit 317628cc (2020-11-16): Fix Servo library!

This fixes #241 and #195.

There were a LOT of things broken about this library...

Turned up a way in which wiring.c may violate expectations of libraries (in fact, it *did* break pulseIn(), just nobody had noticed) when TCD0 was used as millis source, now there's separate macros to convert between microseconds and millisClockCycle and clockCycle. Completed conversion of code to work without the TCA prescaler (this was where most of the problems came from, though not all of them.
presence or absence of used attribute on _timer in ServoTimers has unpredictable (though small [... cut]
closed 2020-11-16 by SpenceKonde