SpenceKonde/megaTinyCore #242: Tons of formatting checks fail....

SpenceKonde, 2020-10-10:
How the hell did this happen?!

per1234, 2020-10-10:
Feel free to ask if you have any specific questions about the way the CI system works.

I'm pretty flakey on the volunteer work these days, but I at least still want to help even if I don't always manage to find the time to make that happen.

SpenceKonde, 2020-10-11:
Thanks - I'm sure I will as I start trying to make things pass - I know my external clock options will have broken the builds... My plan is to get it working here, get it started on DxCore sooner rather than later.... I have to say, I rather like the direction they've been going with AVR. Microchip is announcing so many parts this year - it looks to me like they're trying to replace all the segments of the market that are still using classic AVRs with a rational and consistent new architecture.

per1234, 2020-10-11:
Here are the scripts that can be used to automatically resolve the "General Formatting Checks" failures:
https://github.com/per1234/formatting-checks/tree/master/correction-scripts

---
Here's the command to fix the code formatting:
```
astyle --suffix=none --options=megaavr/extras/ci/arduino-code-style.conf --recursive ./*.pde,*.ino,*.h,*.hpp,*.hh,*.hxx,*.h++,*.cpp,*.cc,*.cxx,*.c++,*.cp,*.c,*.ipp,*.ii,*.ixx,*.inl,*.tpp,*.txx,*.tpl
```
More information:
http://astyle.sourceforge.net/astyle.html

---
Here's the command to fix the misspelled words where possible:
```
codespell --write-changes --skip "./.git,*.gif,*.jpg,*.png" --ignore-words megaavr/extras/ci/codespell-ignore-words-list.txt
```
Note that sometimes there are multiple possible corrections for a misspelled word in the dictionary, and in this case `codespell` is unable to automatically make the correction. The output will look something like this (only an example, this is a spurious detection that I already added to the ignore list when I submitted the PR):
```
.\megaavr\libraries\SoftwareSerial\src\SoftwareSerial.cpp:9: Hart ==> Heart, harm
```
In this case, you do need to manually make the correction.

When there is a false positive, just add the word in lower case to `megaavr/extras/ci/codespell-ignore-words-list.txt`.
More information:
https://github.com/codespell-project/codespell

per1234, 2020-10-11:
>Here are the scripts that can be used to automatically resolve the "General Formatting Checks" failures

If you're on Windows, you can run these scripts from Git Bash, which comes with the Windows version of Git. So they are usable on any operating system.

`astyle` and `codespell` are also cross-platform.

SpenceKonde, 2020-10-13:
Wonderful, thanks!

closed 2020-10-13 by SpenceKonde