SpenceKonde/megaTinyCore #244: can't use megaTinyCore anymore...

ffred, 2020-10-12:
hello,
I'm sure you know it because the changelog is already talking about a 2.1.1 version (not available through update), but in case you don't, I got several problems this morning (French time) with megaTinyCore.
at first I updated to 2.1.0 and couldn't compile for a attiny3216.
get back to 2.0.5 and it was working again.

now nothing is working at all, with this error :
_Skipping contributed index file C:\Users\FLT\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character (']' (code 93)): expected a value
 at [Source: (FileInputStream); line: 1915, column: 12]
 at [Source: (FileInputStream); line: 1915, column: 11] (through reference chain: cc.arduino.contributions.packages.ContributionsIndex["packages"]->java.util.ArrayList[2]->cc.arduino.contributions.packages.ContributedPackage["platforms"]->java.util.ArrayList[24]->cc.arduino.contributions.packages.ContributedPlatform["boards"])
Skipping contributed index file C:\Users\FLT\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character (']' (code 93)): expected a value
 at [Source: (FileInputStream); line: 1915, column: 12]
 at [Source: (FileInputStream); line: 1915, column: 11] (through reference chain: cc.arduino.contributions.packages.ContributionsIndex["packages"]->java.util.ArrayList[2]->cc.arduino.contributions.packages.ContributedPackage["platforms"]->java.util.ArrayList[24]->cc.arduino.contributions.packages.ContributedPlatform["boards"])_

SpenceKonde, 2020-10-12:
Ugh, I screwed something up in the board manager json to fix the 20-pin part problem.

SpenceKonde, 2020-10-12:
Should be fixed now - thanks for reporting. I was unaware of the issue.

closed 2020-10-12 by SpenceKonde

ffred, 2020-10-12:
solved for the attiny414, but still got a problem with attiny3216. got this error :
`Error while uploading: missing 'upload.tool' configuration parameter`