SpenceKonde/megaTinyCore #246: Hardware Serial not working on ATTINY402 on versions 2.1.1 & 2.1.0

alunmorris, 2020-10-12:
Using Serial.begin() etc:
The TxD pin (PA6) goes high but no data comes out. Same using Serial.swap(1).
I noticed the pinout in one of the comments is wrong for TxD, RxD in
https://github.com/SpenceKonde/megaTinyCore/blame/master/megaavr/variants/txy2/pins_arduino.h.
I checked the issue using the example AnalogReadSerial_for_0__1_series_AVR.

I note that the 2018 Atmel datasheet for ATTINY202, 402 has a typo in Table 5-1: SOIC pins 4 & 5 are swapped. 2019 datasheet is correct. Possibly this is the source of the error, though it would not explain why previous versions worked.
Reverted to 2.0.5 and works fine.

Alun

ldalep, 2020-10-12:
Same issue with the Attiny412 here

SpenceKonde, 2020-10-13:
Nope, I just copied it backwards. Will be fixed in 2.1.3, doing the release now.

closed 2020-10-13 by SpenceKonde

SpenceKonde, 2020-10-13:
Now available in board manager, sorry about the messy release...