SpenceKonde/megaTinyCore #247: Error while burning bootloader 2.1.1 - Attiny412

ldalep, 2020-10-12:
Going from 2.0.1 works fine to 2.1.1  gives me an error  - 

Arduino: 1.8.13 (Mac OS X), Board: "ATtiny412/402/212/202, No (saves space), ATtiny412, 20 MHz Internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), Closer to 5v"

/Users/ldalep/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.1/avrdude.conf -v -pattiny412 -cjtag2updi -P/dev/cu.usbserial-AI03K38A -e -Ufuse0:w:0x00:m -Ufuse1:w:0b00000000:m -Ufuse2:w:0x02:m -Ufuse4:w:0x00:m -Ufuse5:w:0b1100{bootloader.resetpinbits}01:m -Ufuse6:w:{bootloader.SYSCFG1}:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.1/avrdude.conf"
         User configuration file is "/Users/ldalep/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-AI03K38A
         Using Programmer              : jtag2updi
avrdude: jtagmkII_getsync(): sign-on command: status -1
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny412
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude: Device signature = 0x1e9223 (probably t412)
avrdude: erasing chip
avrdude: reading input file "0x00"
avrdude: writing fuse0 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse0 written
avrdude: verifying fuse0 memory against 0x00:
avrdude: load data fuse0 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse0 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse0 verified
avrdude: reading input file "0b00000000"
avrdude: writing fuse1 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse1 written
avrdude: verifying fuse1 memory against 0b00000000:
avrdude: load data fuse1 data from input file 0b00000000:
avrdude: input file 0b00000000 contains 1 bytes
avrdude: reading on-chip fuse1 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse1 verified
avrdude: reading input file "0x02"
avrdude: writing fuse2 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse2 written
avrdude: verifying fuse2 memory against 0x02:
avrdude: load data fuse2 data from input file 0x02:
avrdude: input file 0x02 contains 1 bytes
avrdude: reading on-chip fuse2 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse2 verified
avrdude: reading input file "0x00"
avrdude: writing fuse4 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse4 written
avrdude: verifying fuse4 memory against 0x00:
avrdude: load data fuse4 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse4 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse4 verified
avrdude: reading input file "0b1100{bootloader.resetpinbits}01"
avrdude: invalid byte value (0b1100{bootloader.resetpinbits}01) specified for immediate mode
avrdude: read from file '0b1100{bootloader.resetpinbits}01' failed

avrdude done.  Thank you.

Error while burning bootloader.

SpenceKonde, 2020-10-13:
Fixed in 2.1.3, releasing including board manager now.

closed 2020-10-13 by SpenceKonde

SpenceKonde, 2020-10-13:
Now in board manager, sorry for the messy release

ffred, 2020-10-13:
just for info, you keep the 2.1.2 comment for 2.1.3 in board manager :
```
Boards included in this package :
BAD RELEASE - USE 2.1.3 INSTEAD.
```

SpenceKonde, 2020-10-13:
Are you sure it's not showing that because you currently have 2.1.2 installed, so you are being told that the version you hve is bad and you need to use 2.1.3?

SpenceKonde, 2020-10-13:
The board manager json is definitely right as far as I can tell

ffred, 2020-10-14:
I'm on Windows 8.1 on this computer.. screenshot here : [v213](https://i.postimg.cc/KjQ8ZtcV/v213.jpg)

SpenceKonde, 2020-10-14:
That site is not accessible - it pops up a warning, and because the site uses HSTS I can't even tell my browser to ignore it.

SpenceKonde, 2020-10-14:
Yeah, it looks like they forgot to renew their SSL certificate...

SpenceKonde, 2020-10-14:
You can drag and drop files into a github issue to attach a file without having to host it somewhere else

ffred, 2020-10-14:
ok, didn't know I could drag a picture here.. thanks.

[image]