SpenceKonde/megaTinyCore #248: timer problem on attiny414 ?

ffred, 2020-10-13:
hello,
I'm using an attiny414 with 3.3V power supply and select "10MHz internal clock" option.
each time I try to insert a "delay" or "millis" function, my -simple - code seems to stop at that point and don't go further...?
if I remove the delay or millis function, it runs again.

tried with "enabled (default timer)", TCA0, TCB0,.. no change ?

it's a very small board, with no serial output. not easy to "debug" and not much time to search more right now (hope to be able to search more tomorrow).
any idea ? 

(didn't have the problem on my attiny3216 board)

SpenceKonde, 2020-10-13:
You mean a 412? 

Argh... it looks like I managed to fuck up YET ANOTHER THING on the 8-pin parts...

SpenceKonde, 2020-10-13:
Oh, uh, nope, I just forgot about the fact that I was using a first run '412 board with the LED on a different pin.... it seems to be working for me....

SpenceKonde, 2020-10-13:
Please clarify if it's a 414 or 412, and post the code that reproduces it....

ffred, 2020-10-13:
414..
sorry, busy on something important at work now. will give you a code example as soon as I can, but not sure for today.
but that's simple, let's say I set a loop to toggle a pin. with no delay the pin toggle, with a delay(anyvalue), it stops running...

ffred, 2020-10-14:
Hello,
so, I just tried the simplest code I can with a delay to toggle a pin and... it's working !?!
I'm really sorry to take your time about that yesterday. will have to search why my original code is stopping after inserting a delay.

still not much time today. probably faster to start something fresh since I was trying several things at the same time. or at least I will have to clean all that to be sure what the problem is. will let you know if I find my mistake..

**really sorry** again...

_and basic working code :_
```
void setup()
{
    pinMode(PIN_PB2, OUTPUT);
}

void loop(){

        digitalWrite(PIN_PB2, HIGH); 
        delay(400);
  
        digitalWrite(PIN_PB2, LOW); 
        delay(400);        
}
```

closed 2020-10-14 by ffred

ffred, 2020-10-14:
ok, found my problem.. one of the delay was in the wire receiveEvents interruption....
sorry again.

SpenceKonde, 2020-10-18:
Well that would do it...

One can argue about whether the right behavior is to just let it barf, or skip the delay in a totally inappropriate place (there should never be a delay in an ISR, but since there would still be no way to tell the user that happened, I'm not sure either would be any less confusing.... It would be nice if functions could be marked as (don't call from ISR) and have it generate a compile error, since a computer would be perfectly capable of pointing that out, at least if you didn't go to trouble to hide it. A lot of people in Arduino end up writing ISRs without really knowing that they have a whole bunch of additional constraints and complications.... and a brief initial explanation of what they are makes them sound like the way to do all sorts of things.