SpenceKonde/megaTinyCore #249: PWM Hiccups with Timer D on 1-series 20 pin parts

sideswiper22, 2020-10-13:
I've used my ATtiny1616's in an application that maximizes PWM pin usage in a robot car project and I've had quite a few issues when it comes to PWM. At first, the second motor (controlled by pins 10 and 11 via a controller) wouldn't work correctly when an analogWrite() is called.

I found that switching the millis timer to TCA0, which is default to 0-series parts fixed it since I figured that pins 10 and 11's PWM is controlled by the Type D timer. I then had a problem where the PWM signals wouldn't sync with the others or would occasionally produce "hiccups" as the motors ran.

It would sometimes cause a speed imbalance with the other motor which is controlled by PWM pins connected to the Type A timer. At low speeds (at about an analogWrite of 100), the PWM on pins 10 and 11 wasn't able to make the motors run. To fix this, I had to put in a 3-5 millisecond delay on the loop function. It would then work fine, but speed still remained inconsistent, sometimes being slower or faster than the other motor. I've read that you've had a hard time working with the Type D timer since it's a very weird timer, but is this normal behavior? When I first encountered this, I was using version 2.0.5, and I still encounter this issue with the later versions. I haven't tested it with the older versions of the core though.

Just in case this info is needed:
I hooked up 4 PWM pins to my motor controller (I didn't realize at first that 2 would be enough). Motor 1 had it's a and b channels controlled by pins 0 and 1 while Motor 2 was controlled by pins 10 and 11. To get each motor going in one direction, I set one pin to low, while the other transmits a PWM signal (inefficient I know. I could've gotten away with having only one PWM pin while switching the other pin from high to low to for direction. But what's done is done and it's on a PCB. I should've breadboarded it beforehand).

SpenceKonde, 2020-10-17:
Code samples which show this behavior would be useful - I will test some simple cases; the fact that analogWrite in a tight loop wasn't generating output is suspicious; it makes me wonder if my code is always triggering the start/stop behavior.

For turning a PWM channel on or off, (ie, analogWrite(pin,0) or analogWrite(pin,255) or digitalWrite(pin,value)) is expected to produce a very brief glitch in the output. This is an intentional hardware limitation - turning outputs on and off requires that the timer be stopped and disabled, the appropriate register changes (with _PROTECTED_WRITE() - it's a configuration change protection controlled register,. because they are just that scared of the possibility of a wild pointer or something resulting in it getting trashed). It is clearly meant for DC-DC converter and motor control, where a software glitch that results in a bogus pointer writing some random register would mean smoke, fire, sparks and/or pieces of what used to be a BLDC motor flying through the air), and then the timer reenabled. I'll check whether there's a bug there. It sounds like there's a problem with it when TCD0 is used as millis source... 

As I think about it, I feel like there might be a way to, get a constant output without turning off PWM at all... say you're running in one-ramp mode? You want a constant value, so you set CMPxSET higher than CMPxCLR so it will never match, that gives you a constant output (ie, CMPBCLR is 509 as we configure it by default (do we? That's what we should do, I think?), CMPACLR > CMPBCLR (ie, so it gets cleared when it hits CMPBCLR and starts the next timing cycle).... so to get constant value we would set CMP:xSET to a higher value as well, so it would always be in the CMPxCLR state, and we'd poke the INVEN bit of PINnCTRL to choose high or low? I think that might just be crazy enough to work.... 

But that is an enhancement for another day.... I'll check ithough if I'm making an invalid assumption regarding normal TCD0 PWM if it's the millis timer, and also whether there's an unnecessary glitcjh when writing to it frequently...

SpenceKonde, 2020-11-16:
Okay, well, I found the problem with PWM not happening when TCD0 is used for millis (and actually, due to recently removing a "long since removed" define, they didn't work at all). 

Uhm... Long ago, I hadn't known how to make PWM work on the TCD pins while using them for millis. I figured out how to do that in ancient times.... but, somehow that check that disabled PWM on those pins if TCD0 was millis crept back in. Not sure how that happened, or how it went unnoticed for so goddamned long...

SpenceKonde, 2020-11-16:
The timer periods are also now the same too (or they *should* be the same!). TCD0 was off half a tick. Oh - also, if you're running it slower than 16 or 20 MHz, TCD0 runs at a different frequency than the others (it runs at the same frequency as TCA0 does at 16 or 20 at all frequencies dertived from each of those.

closed 2020-11-16 by SpenceKonde

sideswiper22, 2020-12-04:
I can confirm this was fixed and working as of the latest official release.

commit b5e46f17 (2020-11-16): Fix #249, improve TCD0 PWM in general