SpenceKonde/megaTinyCore #250: Bricking my chips

a7hybnj2, 2020-10-15:
I am just starting this issue as a place holder as I dive deeper into this issue. I recently updated my magaTinyCore and after burning bootloader successfully it started bricking my chips. All subsequent uploads or bootloader modifications would fail. I suspect something to do with the updi mode that is being burned.

a7hybnj2, 2020-10-15:
NOTES: So, I did a `git reset --hard ` back to a commit in apr and burning the bootloader appears to work. I will continue to test different hashes and see where it breaks.

SpenceKonde, 2020-10-15:
The configuration of the UPDI pin is controlled by fuse5 (SYSCFG0) - 0b11000101 is no CRC, updi pin as updi, EEPROM retained...

So that configuration should not be breaking things. Previously we would write 0xC7 (0b11000111) there, but bit 1 is unused (so per datasheet it should be getting written 0, not 1), but writing that bit 0 was done before for the optiboot with UPDI pin as UPDI (no autoreset), which also worked...

So this is very mysterious....

a7hybnj2, 2020-10-15:
@SpenceKonde I appreciate you taking a look at this so quickly. Did the functionality of the BODv change since April?

I have determined it doesn't have to do with the UPDI mode but instead the BOD Voltage. In the apr hash I can set any 1.8v 2.6v 4.2v and it doesn't seem to effect anything. The actual BOD Mode is disable/disable.

My programmer dips under 4.2 when programming but when I take the 'bricked' chips on external power they work. So, I guess something with the BOD fuses must have changed at some point.

Anyway, I have this figured out now.

Thanks for this awesome project!

closed 2020-10-15 by a7hybnj2

SpenceKonde, 2020-10-16:
Hmm - it's strange though that this was an issue, since, indeed, I haven't changed anything regarding that (I mean, how you control it has changed - but with it disabled by the old menus and by the new ones got the same fuses).... 

Now, you're probably thinking "how could BOD be a problem with BOD disabled?!" And... well, the reason is that while *programming* BOD is forced on regardless of fuse settings, so that in the event of an undervoltage, it doesn't attempt programming and corrupt the flash. Of course, if BOD is not being used except when turned on for that reason, I don't think there is any situation in which the lowest voltage, 1.8v, would not be appropriate - when UPDI programming is being used, it resets the chip and puts it directly into UPDI programming mode, where it's running at the default speed (2.67 or 3.33 MHz), and if using Optiboot bootloader, likewise, it's running at the slowest speed, if the user disabled BOD and is writing the EEPROM, the undervoltage of 1.8v will protect them from that failure mode, and the window for a "CPU executing instructions incorrectly" failure being relevant seems implauible - if they have BOD off, and are pushing the CPU so hard (in terms of excessive clock speed for Vcc), it's gonna be failing all over the place, and bunged EEPROM writes are the least of their worries. So I think in the next version, I will force the BOD to lowest voltage if disabled/disabled mode is chosen.

SpenceKonde, 2020-10-16:
In any event, my recommendation until then is to set the BOD voltage to 1.8 when running with BOD disabled.