SpenceKonde/megaTinyCore #251: More or Revised SUT Options?

Dlloydev, 2020-10-17:
Can we get more options for Startup Time or a less aggressive minimum setting?

If using the ATtiny 0/1/2 parts as a HV programmer and the target has UPDI in GPIO mode, the HV pulse needs to be applied within 8.8ms after power-up. I find the 1ms option too aggressive and 8ms to close to the allowed limit. I've been measuring the power ramp-up of various circuits and find that it usually takes up to 2ms to get to > 90% of set voltage. In this case 2ms or 4ms are good options but unavailable.

MPLAB options:

[image]

SpenceKonde, 2020-10-17:
Yeah, already doing that for DxCore, planning to bring that back to
megaTinyCore

SpenceKonde, 2020-10-24:
By the way, currently the 8ms option sets it to 4ms. I did a really lousy job on boards.txt in 2.1.0.....

closed 2020-10-27 by SpenceKonde