SpenceKonde/megaTinyCore #253: Adafruit defines (Adafruit_BusIO)

sshwarts, 2020-10-23:
Adafruit is looking for defined(ARDUINO_attinyxy7) ||  defined(ARDUINO_attinyxy6) || defined(ARDUINO_attinyxy4) ||              defined(ARDUINO_attinyxy2).... to set the SPI bit order.

In megaTinyCore, arduino.h these are defined as __AVR_ATtinyxy6__ so the BUS_IO library won't compile. I'd guess there's a definite lack of standardization here and Adafruit went their own way?

SpenceKonde, 2020-10-23:
Hmmmm.... I thought that that was *also* defined, or should be - but I may have just broken that recently. 

Can you link me to the library though? If they are looking for that, they are doing something intensely stupid, I think...

SpenceKonde, 2020-10-23:
nvm found it

SpenceKonde, 2020-10-23:
Yeah, I broke that in 2.1.0....

SpenceKonde, 2020-10-23:
Somewhat embarrassingly, they are specifically testing for my core there - and I bunged the spelling and capitalization and stuff like that. Why they test for the build.board define, instead of something more reasonable is beyond me though....

sshwarts, 2020-10-23:
I’ve given up wondering why Adafruit does some of the things they do….

> 
> 
> Somewhat embarrassingly, they are specifically testing for my core there - and I bunged the spelling and capitalization and stuff like that. Why they test for the build.board define, instead of something more reasonable is beyond me though....

Scott Shwarts
scotts@pobox.com
+1 727 871 0039

SpenceKonde, 2020-10-23:
I've given them a PR for this - seeing as LadyAda cared enough about support to add the old tests, I figure they might be receptive to a PR to fix it :-P

closed 2020-10-23 by SpenceKonde

per1234, 2020-10-23:
My suggestion from when we discussed this on your forum thread:
https://forum.arduino.cc/index.php?topic=623194.msg4768993#msg4768993
>Since it's impossible to know how much other code might have been dependent on the previous macro, the best way forward would be to add a backwards compatibility macro definition here:
https://github.com/SpenceKonde/megaTinyCore/blob/596ca99fb5a491ac0cbe70f12aadc2233d2c2504/megaavr/boards.txt#L197
like so:
>```
>atxy7.build.extra_flags={build.serialevent} {build.millis} -DUARTBAUD{build.uartvoltage} -DARDUINO_attinyxy7
>````

A change to `build.board` should be considered a breaking change, since the `build.board`-based macro is part of the API of a boards platform.

SpenceKonde, 2020-10-23:
Huh - somehow I missed everything before the mention of the BOD settings at the bottom there about this recent post.

I had no idea that the dinky little build.board setting, which I barely paid any attention to was considered like, a super important one

Changed were merged into the Bus_IO library.