SpenceKonde/megaTinyCore #254: Alternate serial issues on 2.1.X builds on the 3217

Pjeran, 2020-10-24:
Loving the megaTinyCore been using it for quite some time.

I am running into an issue where any core version 2.1.1 and above I loose my serial output on the alternate pins.  If I revert back to 2.0.5, even without re-burning the boot loader I get normal serial output.  I have not hooked it up to a scope yet to see if there is activity on the pins.

I use the Serial.swap(1) to accomplish the change to the alternate pins

Was tearing my hair out last night bringing up a new board rev.  Lesson learned only change one thing at a time when bringing up a new system!

SpenceKonde, 2020-10-24:
Hm - like, 2.1.x was a real stinker of a release, and I'm working on a lot of serious breakage... but I dont think I found anything that would explain thius, so if you can get some more information on whats going on, that would help make sure that 2.1.4 which ius coming out within 48 hours at most has this issue fixed...

Usually my upgrades don't break everything x_x

SpenceKonde, 2020-10-24:
And burning bootloader or not should make no difference, as you observe, as the bootloader has not changed.

Pjeran, 2020-10-24:
OK, before I busted out the scope, I did the simple hack of an LED on the TX line and turned the baud down to 9600.

With 2.0.5 - I get a serial output and can see the LED Light and then the data flash over the line.

With 2.1.1 and beyond - the LED does not light and now flashing of data

Seems like there might be something with the pin definitions?  Will do more work on this.

SpenceKonde, 2020-10-24:
Huh, well. that sounds easy, I'll take a look, I'm actually about to test a bunch of other serial stuff on these parts... Oh, and I have to use the alt pins anyway!

SpenceKonde, 2020-10-24:
Yup, alt serial pins are hosed across the board. Still busted in my 2.1.4 version too!

SpenceKonde, 2020-10-26:
Thanks, issue was pretty silly, just wrong pin definitions in pins_arduino (tx and rx reversed). Wound up spending an unreasonable time on it, due to some unexpected behavior - in one case behavior I should have expected from the Arduino core (though this realization allowed me to further reduce the flash used by Serial), and in another case, some rather disturbing hardware behavior, namely:
```
//PORTA.DIR is 0 at the start of this example.
PORTA.DIR=0x10; //same with the other registers in PORTx, eg, DIRSET/etc, and their homologs with OUT
uint8_t test=VPORTA.DIR; //test=0 - wait what?!
```
but:
```
PORTA.DIR=0x10; //same with the other registers in PORTx, eg, DIRSET/etc, and their homologs with OUT
_NOP(); //a single nop instruction - wait 1 clock cycle
uint8_t test=VPORTA.DIR; //test=0x10 like you'd expect!
```

Pjeran, 2020-10-27:
That must have been fun to work out.  Thanks for getting to the bottom of this.

commit 9fa4ee2c (2020-10-27): Fix #254

Oops....
closed 2020-10-27 by SpenceKonde