SpenceKonde/megaTinyCore #255: Get CI compilation of examples working

SpenceKonde, 2020-10-27:
We have almost everything in place to do CI testing! It is targeted for 2.1.5, as we have had quite a run of frantic fixes to patch up all the regressions in 2.1.0 (though, most of these would not have been caught by the CI testing, that is no excuse to not run it - that just means we're due for some that would be!); all that needs to be sorted out now, I think,  are the changes to what we tell CI to do because of the 2.1.x changes, particularly the new clock options...

Dlloydev, 2020-10-28:
Not an issue for successful uploads, but when using 2.1.4 "Upload using Programmer", I'm getting a bunch of warnings (all the same) while compiling:

`<command-line>:0:26: warning: ISO C++11 requires whitespace after the macro name`

SpenceKonde, 2020-10-28:
I cannot reproduce that issue. 

How is this related to the CI testing?

Dlloydev, 2020-10-28:
Ugh ... never mind.
Happened after installing 2.1.4 through boards manager, but still had the manual 2.1.4 in the hardware folder from earlier in the day. Then Java fails to load and Arduino. Now, after a thorough cleanup and re-install its all back to normal.

SpenceKonde, 2020-11-17:
Deferring because between this and the formatting issues, this is kind of a big job

SpenceKonde, 2021-01-08:
This is closer now, but still not there; more important though, the CI scripts need to be adapted to use the different clock sources/speeds and the different millis timer options.

SpenceKonde, 2021-01-27:
This has been sorted out - but it was discovered in the course of this that the CI examples aren't testing with different sources for millis timekeeping, which is.... unfortunate... because that's one of the biggest cause of problems that CI testing could have found!

Closing this, opening new issue

closed 2021-01-27 by SpenceKonde