SpenceKonde/megaTinyCore #256: Serial (hardware) on 3216 @ 3.3V

sshwarts, 2020-10-29:
I'm trying to use serial on a 3216 clocked a 1MHZ internal running at 3.3V. I get only garbage on serial regardless of the baud rate unless I run at 20 MHZ (which is of course not recommended at 3.3V). I'm using the standard pins (PB2, 3). Baud rate calc is set to 3.3V as well. Version 2.1.4.

I've not yet looked with the logic analyzer to see what's happening to the bits.

SpenceKonde, 2020-10-29:
Did you remember to "burn bootloader" when switching between 5/10/20 MHz internal and 1/4/8/16 MHz internal? I thought I had tested this with 2.1.4...

sshwarts, 2020-10-29:
Uh, no I didn't. I am using UPDI programmer (not serial).

If I do choose BurnBootloader in the Arduino IDE, I get an error.

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/scotts/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.4/avrdude.conf"
         User configuration file is "/Users/scotts/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-DA00XIIF
         Using Programmer              : jtag2updi
avrdude: jtagmkII_getsync(): sign-on command: status -1
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny3216
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      32768  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.34s

avrdude: Device signature = 0x1e9521 (probably t3216)
avrdude: erasing chip
avrdude: reading input file "0x00"
avrdude: writing fuse0 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse0 written
avrdude: verifying fuse0 memory against 0x00:
avrdude: load data fuse0 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse0 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse0 verified
avrdude: reading input file "0x00"
avrdude: writing fuse1 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse1 written
avrdude: verifying fuse1 memory against 0x00:
avrdude: load data fuse1 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse1 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse1 verified
avrdude: reading input file "{bootloader.OSCCFG}"
avrdude: invalid byte value ({bootloader.OSCCFG}) specified for immediate mode
avrdude: read from file '{bootloader.OSCCFG}' failed

avrdude done.  Thank you.

Error while burning bootloader.

SpenceKonde, 2020-10-29:
Well,  it *does* say that you need to need to "burn bootloader" to change between those! 

And argh, will do a fix release for that tonight, I just can't seem to do a release without a missing entry or another from boards.txt can I?

SpenceKonde, 2020-10-29:
All boards except the 14-pin non-optiboot definition (the one I tested this all on) were missing that fuse and burning bootloader would fail for non 5/10/20 MHz fuse settings.

commit d20facfc (2020-10-29): Fix #256 - boards.txt missing default OSCCFG fuse

All boards except the 14-pin non-optiboot definition (the one I tested this all on) were missing this entry, and burning bootloader would fail for non 5/10/20 MHz fuse settings.
closed 2020-10-29 by SpenceKonde