SpenceKonde/megaTinyCore #257: Timing problems in Serial and delay()

wschopohl, 2020-10-30:
Let me start by saying thanks for this great project!

I designed a little pcb with an Attiny1614, following your wiring guidelines. I used an Arduino Nano, programmed it in Arduino IDE 1.8.13 with the _jtag2updi_ sketch and than used that to program the Attiny1614. So far everything worked flawlessly. I programmed it without Optiboot at 1MHz internal. I could successfully upload sketches to the Attiny1614 microcontroller that used `analodRead()` and `digitalWrite()` and everything worked as expected.
The problems started, when I tried to use pin 7 as TX to output some debugging information. I used `Serial.begin(19200)` and also tried other baud rates, but only received gibberish in my serial terminal. When I started looking at the output pin with my osciolloscope I noticed, that the timing was off. The baudraute was somewhere near 23896 or something. Afterwards just put the same pin to `HIGH`, `delay(50)` and then back to `LOW`. When looking at the signal it turned out to be rather 40 ms instead of 50 ms.

Does anybody have any idea or clues, what I could be doing wrong? Thanks so much in advance!

SpenceKonde, 2020-10-30:
You did not do burn bootloader to set the fuses to use the correct oscillator to derive the clock, even though the menu literally says right there that you need to! It's using the default 20 MHz clock, but the code thinks it's using the 16 MHz one (because that's what it would be set to if you did "burn bootloader"), is dividing it by 16, and running at 1.25 MHz. 

In 2.1.6 all the fuses except SYSCFG0 (which can brick a tiny if you don't have an HV UPDI programmer) will be set whenever a sketch is uploaded on a non-Optiboot board def so this won't be an issue. It seems that people have been finding this especially confusing lately.

closed 2020-10-30 by SpenceKonde

wschopohl, 2020-10-31:
You are absolutely right! I never realized, that the clock and other menu settings are written when you burn the bootloader - although now I can see, that it says it in the menu. For some reason I thought burning the bootloader is all about putting Optiboot on the uC to program it via serial, which I didn't need... Thanks so much, especially for the quick response!!