SpenceKonde/megaTinyCore #258: I2C broken for default pins

Pjeran, 2020-10-31:
I am running into an issue where any core version 2.1.5  I loose my I2C communication on the default pins. If I revert back to 2.0.5, I can communicate again.  I looks like the system locks up after the first Wire.beginTransmission(address) command, the program does not continue past that command.  I reverted back to 2.0.5 and with an LED on the SCL line I can see data, upgrade to 2.1.5 and nothing.

SpenceKonde, 2020-11-01:
Huh... well that's not good.... 

Can you do a couple of quick tests here? (I've got a lot on my plate atm, and you've got the hardware all wired up, while I'd have to wire up an I2C device, find appropriate libraries for it, get it working on 2.0.5)

Test case 1: 
Do you have an external 4.7~10k resistor between SDA and Vcc, and SCL and Vcc? (ones located on the breakout board for your I2C device count). If not, it is not expected to work. Try it with an external pullup. If that works, we're done. 

Previously we did turn on the internal pullups. In 2.1.x that is removed. I had always wanted to take that out, because it leads people to believe that external pullups are not required - the internal pullups are less than a third as strong as the I2C spec requires. so while there are cases in which it would work, you can also easily find cases where it does not. But I was okay with leaving it in because it wasn't hurting anybody. But in #223 a user complained that this was breaking their code, giving me the excuse to axe what I considered a massive misfeature - though I wonder if maybe I should give in and have it enable the pullup, maybe through a different mechanism that doesn't rely on pinMode.

Test case 2: 
Revert to 2.1.3, repeat tests. 
If this is the problem, it's #235 - though I am surprised that that wound up causing issues in master mode!

Thanks - look forward to hearing results.

Pjeran, 2020-11-01:
Test case 1 solved it - the break out board had 10K pull ups on it, but upon further inspection, they jumper was not soldered.

Note to self - don't assume that even if pull ups are on the board that they are in the circuit.

Sorry for the trouble.

SpenceKonde, 2020-11-01:
Okay phew!

I am going to try to emphasize the importance of using external pullups in
the documentation and see if that prevents people from stumbling over this
(as evidenced by issues); my personal opinion is that unless people just
don't get it when it's done that way, and expect internal pullups to be
used no matter what I say, it's better for it to always fail without them,
than to use internal ones that will work in some - but not all - cases
depending on the details of the configuration.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

Pjeran, 2020-11-02:
I would highlight that even though several sensor boards seem to have pull-ups installed, make sure that they are in circuit.  On one of my temp/humid sensors, there were pull ups, but the jumpers were not closed pulling them into the circuit.  Always check!

thanks again for all your work on this core.

closed 2020-11-03 by Pjeran