SpenceKonde/megaTinyCore #259: Bootloaders broken in 2.1.x?

SpenceKonde, 2020-11-06:
Was reported in #225 that there may be a problem with the included bootloaders in 2.1.x - this is a serious issue for me, as I have shipped several boards pre-loaded with the new bootloaders and reset disabled, and need to address this for customers if I shipped them a board with broken bootloaders and UPDI disabled so they cant reprogram it (I cant either - I still haven't gotten HV UPDI working)

JasXSL, 2020-11-06:
The 8 sec UPDI works fine for me with optiboot. The 1 sec one doesn't work with UPDI. The 2.0 1 sec does work with UPDI. I haven't tested with any of the ones that you need a HV programmer to reset since I don't have one.

SpenceKonde, 2020-11-15:
No, the 8 sec one *doesn't* work in all cases either, though in most cases the defect is hidden there. It's a bug with the handling of the reset flags and entry conditions. We now always clear the reset flags and stash the reset cause in GPIOR0, in order to honor entry conditions even in the face of an ill-mannered user program which does not clear them (which is like 99.9+% of Arduino code - who checks reset cause?! Who even CARES what the reset cause is?!)... have you ever written code that checks the reset cause flags? I haven't except when investigating this kind of thing.... but this can result in Optiboot not honoring the reset conditions, and running when it's not supposed to... so we now clear them and stash them in GPIOR0 before jumping to user code. But this means that they will be zeroed out when the bootloader next runs...

Now, with hardware reset, you don't care, because the DTR line is capacitively coupled to reset and does a hardware reset when you open the serial port, thus setting the flags so they meet the entry condition.... but in any other case, including a freshly bootloaded board, the reset flags likely won't be set. The bootloader will run precisely once on a freshly bootloaded board (when the UPDI reset condition is released by the programmer).... and never again, because it doesn't run on POR and doesn't run when execution skids along empty flash and wraps around, nor when the stub uploaded with the bootloader jumps to 0 - because reset flags are 0, because there wasn't a reset! 

Solution is simple, to check for reset flags being 0... What particularly gets me is that I noticed "gee, why doesn't the bootloader give me the optiboot blink repeatedly" on both DxCore and this... and instead of recognizing that this was a Bad Thing, just went about my merry way and didn't investigate...

SpenceKonde, 2020-11-15:
Fixed with c211403c4009b297f3008c2130f8eb837904d519 note also improvement for ersatz-reset configurations in ebfb0bd013c8ed3237a9aff7a38b3cd38ebeecf1

closed 2020-11-15 by SpenceKonde