SpenceKonde/megaTinyCore #260: warning: ISO C++11 requires whitespace after the macro name (Help me out here!)

SpenceKonde, 2020-11-15:
Current version of megaTinyCore gives this warning all over the place during compilation with warnings enabled. 

This is a serious issue: we must not produce spurious warnings, but I haven't the faintest idea what it is that's making it complain here; I am absolutely concerned that there may be further breakage here that I haven't discovered. Can anyone help me sort this out? I just don't know how to approach a warning that comes up like this without showing me where it's coming from...

```
C:\arduino-1.8.13\arduino-builder -dump-prefs -logger=machine -hardware C:\arduino-1.8.13\hardware -hardware C:\Users\Spence\AppData\Local\Arduino15\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13\tools-builder -tools C:\arduino-1.8.13\hardware\tools\avr -tools C:\Users\Spence\AppData\Local\Arduino15\packages -built-in-libraries C:\arduino-1.8.13\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2o:chip=412,clockopti=20internal,millis=enabled,bodvoltage=1v8,bodmode=disabled,bootloaderuart=alternate,serialevent=no,resetpinopti=UPDI,startuptimeopti=8 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_402729 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\Documents\Arduino\sketches\ErsatzReset\ErsatzReset.ino
C:\arduino-1.8.13\arduino-builder -compile -logger=machine -hardware C:\arduino-1.8.13\hardware -hardware C:\Users\Spence\AppData\Local\Arduino15\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13\tools-builder -tools C:\arduino-1.8.13\hardware\tools\avr -tools C:\Users\Spence\AppData\Local\Arduino15\packages -built-in-libraries C:\arduino-1.8.13\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy2o:chip=412,clockopti=20internal,millis=enabled,bodvoltage=1v8,bodmode=disabled,bootloaderuart=alternate,serialevent=no,resetpinopti=UPDI,startuptimeopti=8 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_402729 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\Documents\Arduino\sketches\ErsatzReset\ErsatzReset.ino
Using board 'atxy2o' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr
Using core 'megatinycore' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr
Detecting libraries used...
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy2" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\sketch\\ErsatzReset.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy2" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\sketch\\ErsatzReset.ino.cpp" -o "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\arduino-1.8.13\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy2" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\sketch\\ErsatzReset.ino.cpp" -o "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\sketch\\ErsatzReset.ino.cpp.o"
<command-line>:0:9: warning: ISO C++11 requires whitespace after the macro name
Compiling libraries...
Compiling core...
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy2" "C:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore\\wiring_pulse.S" -o "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\core\\wiring_pulse.S.o"
<command-line>:0:9: warning: missing whitespace after the macro name
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy2" "C:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore\\hooks.c" -o "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729\\core\\hooks.c.o"
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny412 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny412 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.0\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=0UL -DMEGATINYCORE_RELEASED=0 -DARDUINO_attinyxy2 -DNOSERIALEVENT -DMILLIS_USE_TIMERD0 -DUARTBAUD-V "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Ard [... cut ...]
cmd /C "C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729/ErsatzReset.ino.elf" > "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729/ErsatzReset.ino.lst"
"C:\\arduino-1.8.13\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_402729/ErsatzReset.ino.elf"
Sketch uses 842 bytes (23%) of program storage space. Maximum is 3584 bytes.
Global variables use 10 bytes (3%) of dynamic memory, leaving 246 bytes for local variables. Maximum is 256 bytes.
```

Dlloydev, 2020-11-15:
Well, that took a while to reproduce! (#255)

For me, I got similar spurious warnings and then posted my comment. The next day I found that this was a prelude to the Arduino IDE totally failing to load (reporting some problem with java). Crazy thing is that the "C:\Users\xxxxx\AppData\Local\Arduino15" folder had grown in size to greater than 1.5GB which seemed really excessive! Deleting this folder and an uninstall/reinstall of Arduino resolved the problems. Seems like an Arduino IDE problem to me.

Note: 
I use compiler warnings set to "More"
More than 2 weeks later, the size of my "C:\Users\xxxxx\AppData\Local\Arduino15" folder is just over 600MB and everything is working fine.

SpenceKonde, 2020-11-15:
Ugghhhhhh lovely....

SpenceKonde, 2020-11-15:
I'm not seeing the bloat of the arduino15 folder though, hmm... but good that I know what to do if Arduino IDE suddenly fails on me. So nobody else is getting these spurious warnings with the latest version of the core?

per1234, 2020-11-15:
>So nobody else is getting these spurious warnings with the latest version of the core?

I am. It's only with specific boards selected. You can see the cause here: `-DUARTBAUD-V`

Which comes from here:
https://github.com/SpenceKonde/megaTinyCore/blob/ebfb0bd013c8ed3237a9aff7a38b3cd38ebeecf1/megaavr/boards.txt#L1834

Same problem here:
https://github.com/SpenceKonde/megaTinyCore/blob/ebfb0bd013c8ed3237a9aff7a38b3cd38ebeecf1/megaavr/boards.txt#L864

>a warning that comes up like this without showing me where it's coming from...

It does say where it's coming from, the command line:
>```
><command-line>:0:9: warning: ISO C99 requires whitespace after the macro name
>```

SpenceKonde, 2020-11-15:
AAAHAAA! 

Thanks, I've got it!

SpenceKonde, 2020-11-15:
- is not legal in the name of a macro, so it was coming from the -DUARTBAUD-V on the xy2 parts (we don't do UART baud rate correction based on compile-time-reported baud rate for those parts, since the calculations are too flash-intensive and it doesn't matter in typical use cases - one could argue that the UART baud voltage correction feature should be dropped from megaTinyCore entirely); factory cal is easily good enough for UART communication with a serial adapter without it.

closed 2020-11-15 by SpenceKonde