SpenceKonde/megaTinyCore #261: Simple sketch pwm on attiny412 glitch blink

yuranikspb, 2020-11-17:
```
byte output_pwm_pin = 3;
void setup() {
  // put your setup code here, to run once:
 pinMode(output_pwm_pin, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
for(int i=0 ;i<255;i++)
{
analogWrite(output_pwm_pin,i);  ///blink at about 200 pwm
delay(20);
}
delay(2000);

for(int i=255 ;i>0;i--)
{
analogWrite(output_pwm_pin,i); /// not blink
delay(20);
}
delay(2000);
}
```
i am connect led with resistor to pin 3 
when you look at Led it sometimes blinks until pwm is not maximum
that is, in the process of changing the duty cicle, as  pwm signal is not correct

this sketch uploaded to arduino UNO work good led not blink
not hardware problem

sorry for my bad english/

SpenceKonde, 2020-11-19:
Hi - thanks for the report. 

I cannot reproduce this; does it happen with the latest version from github? (not the release; I have fixed some things that might be related to this, and those fixes are not yet in a release)

What do you have selected from Tools -> millis/micros menu? Does changing that selection make a difference?

Thanks!

yuranikspb, 2020-11-19:
[image]

> have fixed some things that might be related to this, and those fixes are not yet in a release
i will try this.

yuranikspb, 2020-11-19:
`have fixed some things that might be related to this, and those fixes are not yet in a release`

i tried it didn't help

ldalep, 2020-11-20:
Just out of curiosity did you try using TCB0 under the millis()/micros() pulldown

SpenceKonde, 2020-11-20:
Also wondering about trying other timers - idalep saw glitches in PWM on the t412 too, but only with default timer (TCD0) used as millis clock source; with TCB0, it did not occur for him. But his code was hard to set up to reproduce. Your code seems simple.... except I don't see the glitch....

ldalep, 2020-11-20:
Mine was on the 1604 - I have new boards that work good with the the TCB0 (better bypassing... ) - I will go back and verify that the default is still broke.

SpenceKonde, 2020-11-20:
Oh wait, you are not the other case I was thinking of are you....

yuranikspb, 2020-11-20:
Change default timer does not help(no diference)
There are a dependence on the clock frequency:
1: 20 mhz - often there is no glitch on the first cycle all other cycles have (from 0 pwm to 255, at about 200pwm blink)
2: 1 mhz -glitch on the first cycle and all other cycles have (from 0 pwm to 255) at about 50 pwm and at about 180pwm. And blink much more visible.

in all variants from 255 to 0 no glitch

ldalep, 2020-11-20:
Hi All - I set up my board and downloaded the code from above - I confirm using the red portion of RGB Led and my board  (412) and I am seeing the glitch that yuranikspb discusses.  Unfortunately it is slight and using the various timers made no difference in this case.  Also sometimes it appears to skip cycles giving no glitch - I guess I could drag out the scope...

SpenceKonde, 2020-11-27:
Yup, I can see it in the video... I'm not sure why I can't reproduce it, and I find that fact rather disturbing...

But I think I know what's happening... the clues are that it goes dark, not bright (I'm going to guess for 2.04ms...), and only when raising the PWM duty cycle (a clue I had missed originally)

I'll bet it's just as bad at 8 MHz as it is at 1 MHz, and happens just as much at 16 MHz but is harder to see (because the glitch lasts half as long). It's hardest to see at 20 MHz because the glitch there would last only 816us). 

Does putting this before the analogWrite() fix it? 

`while ( ( (uint8_t)(TCA0.SPLIT.LCNT-(i-1)) ) < 2 );`

ldalep, 2020-11-28:
I see no difference in my setup....

yuranikspb, 2020-12-01:
while ( ( (uint8_t)(TCA0.SPLIT.LCNT-(i-1)) ) < 2 );
first  cycles good
other glitch (20mhz)

SpenceKonde, 2021-01-27:
Unsure if it is possible to improve this further without major changes; 

that said, if you were to take over TCA0, switch it from split mode into normal mode, and write your own replacement for analogWrite() (see my examples of taking over TCA0 in the extras folder and the examples under the megaTinyCore "library") and then write into the TCA0.NORMAL.CMPnBUF registers instead of the CMPn registers... that would 100% get rid of the glitch. 

The problem is that buffering isn't supported in split mode. You would lose 1 PWM channel ( on the tiny412, with only 4 PWM channels available, many people would be willing to sacrifice one to get glitch-free PWM on the others...... The same is not true on the larger chips, where you'd sacrifice three PWM channels - and let's face it - we're in Arduino-land, and most people care more about how many 8-bit PWM channels they can analogWrite() on vs how fancy the PWM they can generate on half as many pins is). But - I don't have time to implement something like that. Too much to do. Not enough time!

SpenceKonde, 2021-01-27:
Anyway, closing this - I think I've done what I can here. (I never *was* able to generate something that I could see on the bench here...)

closed 2021-01-27 by SpenceKonde