SpenceKonde/megaTinyCore #262: TXY4 Pin map is incorrect in txy4\pins_arduino.h

mechatroniks-git, 2020-11-20:
These parts do not have A10 or A11, and the map is missing B0-B3.

SpenceKonde, 2020-11-20:
It is correct. Please see the "ways to refer to pins" section of the documentation. ....

The PIN_An defines (where n is a number), following the convention set on the official Arduino core, refer to the digital pin corresponding to a given ADC (analog channel) pin number, these correspond to the "analog pins", the An constants - in megaTinyCore (and DxCore - ATTinyCore is what might be technically termed a "shitshow" because I was inexperienced when I started maintaining it, and it needs significant work to clean up things related to this).

Because I hate the idea of "analog" and "digital" pins, I actually have everything defined as constants back to the digital pin numbers, so there is no separate set of pin numbers, the An constants are just dummy defines for compatibility with example/etc code which uses the An notation. The digital pin referring to pin Pxn where x is the letter corresponding to a port, and n is a number of a bit within that port, are named PIN_**P**xn to prevent conflicts with the conventions set by the official Arduino core. Just below where you changed the defines to make them inconsistent with the documentation, are the #defines for the PIN_Pxn notation, which is what you were looking for

closed 2020-11-20 by SpenceKonde

mechatroniks-git, 2020-11-23:
Thanks for the heads up, I agree with you, the nomenclature is confusing.