SpenceKonde/megaTinyCore #265: Serial.print or other Serial related stuff not working.

IMFROST, 2020-11-23:


SpenceKonde, 2020-11-23:
Issue is unrelated to jtag2updi (though you should be using a 470 ohm resistor - 4.7 is pushing the upper bounds of values that will work; if I still have any references suggesting 4.7k from back before I learned the error of my ways this spring, please let me know) or how you are programming the part. 

Serial is working for me... I have been actively using Serial on these parts recently without issue. You sure your serial adapter is connected correctly? I still get this wrong all the time if I don't have a 6-pin FTDI header to make it literally impossible to plug them in wrong. RX will be on PB3 (Arduino pin 4 / PIN_PB3, physical pin 6), and TX on PB2 (Arduino pin 5/PIN_PB2, physical pin 7), and ground of serial adapter should be tied to attiny's ground.

IMFROST, 2020-11-24:
Let me start by saying THANKS A LOT. 
I am a big fan of yours.
Second, Please excuse me but I forgot to mention that I am a total beginner and I just assumed that you did something with your megatinycore to make the Serial communication work with the UPDI connection. (I did spent a few days on Youtube and google to see if that was true or not). But did not find any reference that would indicate if it was true or not.
Some people that did show Serial communication in their videos or articles did not mention that they were using USB to Serial Adapters. So I just assumed that since the UPDI is able to program the Attiny then it can also communicate as well.

That was why it was not working.

So now that you mentioned about the adapter. I soldered wires to the TX and RX pins of the attiny IC and hooked it up to an FTDI adapter. And it works flawlessly.

Again I am sorry that I did not do enough research about UPDI and UART before opening an issue here.
[image]
[image]

IMFROST, 2020-11-24:
About the 4.7K resistor. 
Now that I think about it, all of your new articles does not have the 4.7K resistor so I just inserted it in there because I remembered from your previous articles that you should have 4.7K resistor.
I will replace that with 470 right away.
Although its probably worth mentioning that I have NEVER had a problem with 4.7K even sometime used 5.6K.
I have been following your work for almost a year now.
I am still a noob but you have to start somewhere I guess.

closed 2020-11-25 by SpenceKonde