SpenceKonde/megaTinyCore #266: Compilation fails Attiny 402

DerPicknicker, 2020-11-26:
Hello, 

I can’t compile that code which you can find in the attachment. Any suggestions? Please rename the File to .ino.. 
Also in the attachment is the full error Log.

[BasicUsage Kopie.ino.txt](https://github.com/SpenceKonde/megaTinyCore/files/5603347/BasicUsage.Kopie.ino.txt)

Thank you very much. 
[error Log.txt](https://github.com/SpenceKonde/megaTinyCore/files/5603359/error.Log.txt)

SpenceKonde, 2020-11-26:
You dont have enough flash. The PCREL error is often reported on the 4k
parts because it is unable to calculate the rjmp addresses, so it fails
before it gets as far as generating a binary that doesnt fit.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-11-26:
Make sure you are using th version of the core from github, not the board
manager version, I saved a few hundred bytes from Serial but havent rolled
up a release because working on some other fixes and enhancements

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

DerPicknicker, 2020-11-26:
> Make sure you are using th version of the core from github, not the board
> manager version, I saved a few hundred bytes from Serial but havent rolled
> up a release because working on some other fixes and enhancements
> 
> Spence Konde
> Azzy’S Electronics
> 
> New products! Check them out at tindie.com/stores/DrAzzy
> GitHub: github.com/SpenceKonde
> ATTinyCore: Arduino support for almost every ATTiny microcontroller
> Contact: spencekonde@gmail.com
> 
> 
> 
> > Hello,
> >
> > I can’t compile that code which you can find in the attachment. Any
> > suggestions? Please rename the File to .ino..
> > Also in the attachment is the full error Log.
> >
> > BasicUsage Kopie.ino.txt
> >
> > error Log.txt
> >
> > Thank you very much.

Thanks for the quick response. I will do it soon as possible. The code is too big for 4kb and that's why it's failed, right? 

Can I set the lock Bits with your Core for the ide. I mean to protect the source code.

SpenceKonde, 2020-11-27:
The SoftwareSerial library, in general, is an abomination... 

megaTinyCore does not support setting the lockbits. On (non-optiboot) configurations, while it could be done, it is complicated by the need for changes to the upload process as well (for example, we couldn't "verify" the upload...). My position on this is the same as it always has been: use avrdude manually to set the lockbits after (or while) uploading the code. When going into production, you wouldn't be using the IDE to upload anyway (you'd be manually uploading firmware by calling avrdude or similar - or for larger runs, having your supplier program the parts for you). 
`C:\arduino-1.8.13\hardware\tools\avr/bin/avrdude -CC:\Users\Quimet\Documents\Arduino\hardware\megaTinyCore\megaavr/avrdude.conf -v -pattiny402 -cjtag2updi -PCOM3 -Ulock:w:0x00:m `

Also worth noting: the source code can never be extracted from the device - it isn't there - though without the lockbits set, they could still extract the binary, so they could "xerox" the chip, but not modify the code (at least not easily - I modified the mEDBG binary used by microUPDI to remove the dependency on AREF - which was an extremely favorable case, to put it mildly).

Another thing - pins 10 and 11 for the software serial, huh? I suppose you hadn't gotten far enough to have to choose pins which existed anyway, but it's IMO rather horrifying that the SoftwareSerial library lets you compile that... I ought to put in a compile-time check for that, but on the other hand, I don't really want to encourage use of SoftwareSerial....
Probably ought to throw together a half-decent software serial implementation for megaTinyCore that's reasonably space efficient; I think writing a bitbang serial implementation in inline assembly is sort of a rite of passage... it's not particularly hard, and I've already modified two implementations of one...  but I don't need to add more tasks to my to-do list right now!

Hardware serial (Serial) is on pins 0 and 1. See https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x02.md

The issue you'd have with hardware serial is that in the sketch you sent, it's  already using the hardware serial port to send readings to the computer... But if you don't plan to do that, yeah, that's no problem (I can tell that that's initial test code - probably the example code from the library for that CO2 sensor).

DerPicknicker, 2020-11-27:
Thank you for your helpful answer! 

I don’t need the Communication between computer and attiny if the code works. It was only for calculating the storage usage. One example of the softwareSerial libary fills around 85% of the Space...
If I write in the Setup(): 
Serial.begin(9600);

Then it refers to the Pins 0,1 -> right? 

I don’t need to specify the pins? 

After I did some research, the libary of this
 mh-z19b Sensor needs so much space and offers function which I don’t need. So I will write the Code directly for the sensor and this saves a lot of space. 

Have a great Weekend!

SpenceKonde, 2020-11-27:
Yup, you do not need to specify the pins. Note that the pins are the other way around compared to Uno et al - 0 is TX, 1 is RX (on atmega328p-based Arduinos, pin 0 is RX, pin 1 is TX).

If you need to (because you need some functionality of those pins for something else), you can move it to the alternate pins (shown on the pinout chart) with `Serial.swap(1);` though you do need to do that *before* calling `Serial.begin()` .

SpenceKonde, 2020-11-27:
And yeah, bloated libraries suffering from "kitchen sinkism" are endemic to Arduino-land. Bad code in general is - but that's what you get when you make it so easy to set up a development environment that an average household cat can do it.... Hell, they've got schools using Arduino as kids' first introduction to programming - and by all accounts, it's more successful than most pre-college programming instruction... *Embedded C++ as a first introduction to programming* - who would have thought that would work out?!

closed 2020-11-27 by SpenceKonde