SpenceKonde/megaTinyCore #267: Adafruit_PWMServoDriver  on core version higher then 2.0.5 not work chip attiny1614 bootloader

yuranikspb, 2020-11-26:
there are test sketch

```
#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>

const byte output_led_zero = 10;

Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

void setup() {
  // put your setup code here, to run once:
 // Serial.begin(115200);
 // pinMode(pin_input_tail, INPUT);
 // pinMode(pin_input_stop, INPUT);
  pinMode(output_led_zero, OUTPUT);
  pwm.begin();
  // In theory the internal oscillator is 25MHz but it really isn't
  // that precise. You can 'calibrate' by tweaking this number till
  // you get the frequency you're expecting!
  pwm.setOscillatorFrequency(27000000);  // The int.osc. is closer to 27MHz
  pwm.setPWMFreq(500);  // This is the maximum PWM frequency

  // if you want to really speed stuff up, you can go into 'fast 400khz I2C' mode
  // some i2c devices dont like this so much so if you're sharing the bus, watch
  // out for this!
  Wire.setClock(400000);

}

void loop() {

analogWrite(output_led_zero, 255);
delay(500);
analogWrite(output_led_zero, 0);
delay(500);
}
```
pin 10 connected to led
in version 2.0.5 led blinking ok

in version higher then 2.0.5  not blinking

iansexton, 2020-11-27:
I've had very little success with any Adafruit I2C libraries.

SpenceKonde, 2020-11-27:
Are you sure this is related to the Adafruit_PWMServoDriver? Quick look at the code for that library doesn't look like there's anything about it that would cause any problems here. 

Do you have external pullup resistors in place on the SCL and SDA lines? One change with 2.1.x is that I no longer enable the internal pullups for the I2C lines - they are unreliable if they're the only pullup, and enabling them wound up causing some problems in other contexts. The user argued that we shouldn't be touching them at all, since external ones should be in use. I agreed on the grounds that it's better to fail 100% of the time than to use a "hack" that works sometimes, but not always (for example, it might work with one I2C device with short wires, but then fail with multiple, or fail if the wires were too long, or if the I2C clock was made faster, and so on). 

If this is the case, the sketch, in fact, would be hanging when it first tries to talk to the I2C device, and got no response. 

Many breakout boards for I2C devices include pullup resistors for SCL and SDA - however there may be a jumper or solder bridge required to enable them.

yuranikspb, 2020-11-27:
> Are you sure this is related to the Adafruit_PWMServoDriver? Quick look at the code for that library doesn't look like there's anything about it that would cause any problems here.
>
> Do you have [...]

This is my mistake, I did not think that I could forget about the pull-up resistors, the more it is written in the change log about this.
"Wire no longer tries turning on pullups; this caused problems when switching between master and slave, and the internal pullups just aren't strong enough for this! (#223)"

Sleepless night

closed 2020-11-27 by yuranikspb