SpenceKonde/megaTinyCore #273: analogWrite 3216 pins 10, and 9 last github version  analogwrite(255) off pwm

yuranikspb, 2020-12-03:
download last github version 
analogwrite on pin is working but?
pwm on 9  if analogwrite(255) off pwm
pwm on 10 pin work good

SpenceKonde, 2020-12-03:
Can you clarify exactly what behavior you are seeing?

analogWrite(pin,255) should turn off PWM on that pin, because you are instructing it to do 100% duty cycle, ie, not PWM.

yuranikspb, 2020-12-03:
analogWrite 254 (on 9 pin led on )
analogWrite 255 (on 9 pin led off )

analogWrite 254 (on 10 pin led on )
analogWrite 255 (on 10 pin led on )

SpenceKonde, 2020-12-03:
Oh eeeew! That's no good!

Pins 16, 0, 1, 7, and 8 show the same behavior too, I imagine?

closed 2020-12-03 by SpenceKonde

yuranikspb, 2020-12-03:
yes , and glitch like https://github.com/SpenceKonde/megaTinyCore/issues/271 on rising duty cicles

SpenceKonde, 2020-12-03:
Fix for #273 is in, that was a silly mistake.

#271 was specific to PWM going from 0 to 1, and it should be fixed, do you mean #261 ?

yuranikspb, 2020-12-03:
yes i mean 261

SpenceKonde, 2020-12-04:
PWM on TCA pins should again work in latest github version... my fix earlier this morning put the changes in the wrong place, breaking it completely instead of fixing it.

commit f9602351 (2020-12-03): Oh duh... (fix #273)

|=, &=~ - all the same thing right?