SpenceKonde/megaTinyCore #274: Question about BOD

JasXSL, 2020-12-04:
I need to use BOD in a circuit, and I'm looking at the options in the arduino menu. Each option is followed by ```(n mhz or less)```. Does that mean the clock frequency of the attiny can't be higher than that? Or is that the frequency it's testing at?

The reason I'm asking is I would like to use 2.7V BOD with a 20mhz attiny. I'm using optiboot if that makes any difference. I didn't see anything in the documentation about it.

SpenceKonde, 2020-12-04:
That is the frequency at which the attiny is guaranteed (Microchip's word, not mine) to operate at when above that BOD threshold across the -40 - 105C operating temperature range described in the datasheet.. See a bit more summarizing at https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/SpeedGrades.md - in addition to that, I do mention in the clock speed section that 20/16 MHz is typical for 5v systems, and 10/8 MHz is typical for 3.3v systems. 

Arguably, this is a datasheet question, not an Arduino core documentation level question - but their answer is that the ATtiny is not within manufacturer specification at 2.7V if you are running at 20 MHz - but they are speaking of the entire temperature range and with a reliability expectation appropriate for these being used in, for example, on-board control systems in vehicles, where a system crash could mean a literal system crash, with attendant damage to property, loss of life, and - even more importantly - liability for Microchip. So, of course they're going to spec the chips very conservatively.

Under less demanding operating conditions encountered in hobby electronics - near room temperature, not outdoors at night in the freezing wastes of Siberia nor sitting in the sun at high noon in the Libyan desert, and input for the digital power supply isn't coming from a notoriously (electrically) automotive power network, and that (like anything else built on Arduino code) it is not being used in a life-safety-critical application, I'd be surprised if it didn't work.

SpenceKonde, 2020-12-04:
To be clear, the clock speed has no bearing on the behavior of the BOD - 2.7V BOD just doesn't guarantee that the chip will work at 20 MHz.

closed 2020-12-04 by SpenceKonde

JasXSL, 2020-12-04:
Thanks for the clarification 👍