SpenceKonde/megaTinyCore #275: Compile issue using RTC on 1604  2.1.5

ldalep, 2020-12-08:
My thoughts were to use an external 32Khz clock and  use the millis() function to do accurate time keeping for battery app.  If i read the docs correct this could be a crystal with caps on the TX and RX pins of the 1604.   Move the TX, RX to optional pins.   I am having problems understanding which pull down to use because they seem duplicated twice under (millis()/micros()).  Trying any of them using the RTC seems to give me the following compile error

Ultimately I would like to use the power saving part but am wondering does the external 32Khz oscillator defeat the purpose

Arduino: 1.8.13 (Mac OS X), Board: "ATtiny1614/1604/814/804/414/404/214/204, No (saves space), ATtiny1604, 20 MHz Internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, RTC w/w32.768 kHz ext. crystal (1/2-series only, no micros), UPDI (pick this unless you have an HV UPDI programmer!), 8ms, Closer to 5v"

                 from /Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring_private.h:26,
                 from /Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:23:
/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c: In function 'init_millis':
/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:508:28: error: 'CLKCTRL_t {aka struct CLKCTRL_struct}' has no member named 'XOSC32KCTRLA'; did you mean 'OSC32KCTRLA'?
   _PROTECTED_WRITE(CLKCTRL.XOSC32KCTRLA, CLKCTRL_CSUT_16K_gc | CLKCTRL_ENABLE_bm | CLKCTRL_RUNSTDBY_bm);
                            ^
In file included from /Users/ldalep/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/io.h:722:0,
                 from /Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring_private.h:26,
                 from /Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:23:
/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:508:42: error: 'CLKCTRL_CSUT_16K_gc' undeclared (first use in this function); did you mean 'CLKCTRL_PDIV_16X_gc'?
   _PROTECTED_WRITE(CLKCTRL.XOSC32KCTRLA, CLKCTRL_CSUT_16K_gc | CLKCTRL_ENABLE_bm | CLKCTRL_RUNSTDBY_bm);
                                          ^
/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:508:42: note: each undeclared identifier is reported only once for each function it appears in
/Users/ldalep/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.1.5/cores/megatinycore/wiring.c:508:64: error: 'CLKCTRL_ENABLE_bm' undeclared (first use in this function); did you mean 'CCL_ENABLE_bm'?
   _PROTECTED_WRITE(CLKCTRL.XOSC32KCTRLA, CLKCTRL_CSUT_16K_gc | CLKCTRL_ENABLE_bm | CLKCTRL_RUNSTDBY_bm);
                                                                ^
exit status 1
Error compiling for board ATtiny1614/1604/814/804/414/404/214/204.

ldalep, 2020-12-10:
Sigh .... in reading the closed issues this isn’t going to do what I want .... definitely not a biggy

SpenceKonde, 2020-12-10:
Wait for the magic sleep library yo ;-)

SpenceKonde, 2020-12-10:
Oh, didn't you read the option? The external 32.768kHz timebase options are only available on the 1/2-series parts. You are using a 0-series part. The hardware does not support this functionality.

closed 2020-12-10 by SpenceKonde

ldalep, 2020-12-10:
Thank You - somewhat clearer..... It does compile if I choose 1614  but does not on the RTC choice  ( I assume this is internal) .... Also I have duplicate choices for the external crystal.  Not sure what happened there

SpenceKonde, 2020-12-10:
The options are Ext. Oscillator and Ext. Crystal. 

A crystal connects with two pins and is just a passive piece of precision shaped quartz in a can. An oscillator connects with one pin and requires power connection as well, and also costs a heck of a lot more :-P but frequently is more accurate.

ldalep, 2020-12-10:
Great.... Got that now but what is simply RTC - That does not compile with 1614

SpenceKonde, 2020-12-10:
Fixing that, trivial blunder - RTC without anything external uses the internal (low accuracy) 32 KHz ULP oscillator.

(and to answer a question well above, a watch cerystal can dissipate a minicule amount of power :-P

ldalep, 2020-12-10:
Sorry to bug you - RTC should work on series 0 as well as 1

SpenceKonde, 2020-12-10:
Yes, correct. But the way it's written currently, it won't even compile
anywhere :-P

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

ldalep, 2020-12-10:
Thank you for the hard work and the fast turn :)

SpenceKonde, 2020-12-11:
fix for the RTC millis issue is in github.

ldalep, 2020-12-11:
Works great  verified the crystal in hardware can't wait for the low power libraries - this gets me to the consumer / battery are that is needed