SpenceKonde/megaTinyCore #280: Help required: Compiling works - Flashing not

DerPicknicker, 2020-12-13:
I have tried to flash the ATTINY404. But it did not work and always ended with an error message. I have then subsequently installed the capacitor and the resistor. 
However, this also did not help to flash the ATTINY. 

Have I damaged the ATTINY by initially missing resistor? How can I find out where the error lies? 

I have attached a file for you, in which the error message is contained. 

In the meantime I will test another Tiny to exclude a defect. 

Best regards 
[errorlog.txt](https://github.com/SpenceKonde/megaTinyCore/files/5684930/errorlog.txt)

SpenceKonde, 2020-12-16:
```
Beim Hochladen des Sketches ist ein Fehler aufgetreten
Reading | avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude: error reading signature data for part "ATtiny404", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE
avrdude done.  Thank you.
```
Relevant part quoted above. 

Likely incorrect connections and/or problem with jtag2updi - is your jtag2updi programmer known to actually work? Make sure you pulled the most recent version from my repo - there was a nasty bug in there that made it completely non-functional in most cases for ;like a month until it was fixed some time in mid-late november...

SpenceKonde, 2020-12-16:
I for one am really really looking forward to 2.2.0 when I will no longer have to recommend jtag2updi, which has been nothing but pain and misery.

Dlloydev, 2020-12-17:
Just some thoughts...

If the target is on an SOIC to DIP adapter, make sure to have a 0.1µF bypass cap installed as close as possible to the power pins. Try a series resistance (270 ohm to 2.2K) in the UPDI signal (not 4.7K) and keep the leads as short as possible.

Note that if you don't have a HV Programmer and the UPDI pin is programmed with any other selection (Reset or GPIO) then this will brick the chip. I've read that others have recovered older iteration MCUs of this type by breifly touching 12V to the PA0 pin, then successfully re-programming.

When you say "flash", do you mean "Upload Using Programmer" or "Burn Bootloader"? Note that some configuration changes require you to "Burn Bootloader" first.

DerPicknicker, 2020-12-21:
Sorry for my late reply - I was so busy ... 

I found the Problem it was caused by the wiring .. the 5V Pin was not able to supply stable 5V ... I had big voltage drops to 1,2v .. 
In conclusion: 
Double check the wiring!! 

What do you mean with the 2.2.0 Version could you explain that? 

Stay healthy..

SpenceKonde, 2020-12-27:
The upcoming 2.2.0 release will add support for programming with just a serial adapter and 4.7k (approx) resistor, like pyupdi.

closed 2020-12-27 by SpenceKonde