SpenceKonde/megaTinyCore #281: pulseIn measurement incorrect at 10 MHz clock with ATtiny412

Fins1, 2020-12-14:
I recently moved code from an ATtiny85 to an ATtiny412 custom board.  When compiling with the clock set to 20 MHz, the pulseIn function reads correctly, within 1% for a 1 ms pulse and about 4 % low for a 2 ms pulse.  Close enough for my application.  However, when compiling with a 10 MHz clock, the pulseIn function reads a factor of 2 low.  I need to run the 412 at 10 MHz since I am using 3.3 V over a moderately wide temperature range.  The code in question is simple:

    throttle_PW = pulseIn(THRpin, HIGH, 50000)

Other timing functions, such as delay and tone produce correct results at both 20 and 10 MHz clocks.  For now, I just added a "2 * " to the above and it works correctly at 10 MHz.  I have not verified other clock frequencies, but suspect this could be a problem for other clock frequencies as well.

SpenceKonde, 2020-12-15:
Oh, I noticed and fixed this last week while working on Servo (or was it PWM? It was part of a Servo related commit, but might have noticed while working on PWM); you'll find that it works if you select TCA0 or one of the TCBs for millis/micros timing. We run TCD0 off the *unprescaled* internal oscillator; this prevents loss of micros accuracy as clock speed is lowered, and, when it's also used for PWM, keeps the PWM frequency from falling, too... it was first implemented nearly a year ago, and TCD0 was made the default option on the 8 and 14 pins around the dawn of the quarantimes, and for all parts a few months after that. But when I did that, I hadn't realized that the macros involved in converting between microseconds and clock cycles were used elsewhere (how I managed to not realize that, I couldn't tell you, heh)... I noticed last week or the week before that this was one of the things (among many) that broke servo, and fixed it then (we were always returning unprescaled clock cycles, but everything other than millis that counts clocks wants prescaled clocks per uS)....

SpenceKonde, 2020-12-15:
Fix will be in next released version (2.2.0)

closed 2020-12-15 by SpenceKonde