SpenceKonde/megaTinyCore #282: SERNUM register not defined

jashsheth, 2020-12-15:
Hi community,
I am trying to access the 10bit unique serial number in ATTINY1614 , which resides in SERNUMn registers.
When simply trying to do a register read, I am not able to do it as it says that this register is not defined in the scope.
Please help

Thanks

------------------------------------------------------------------------------------
Attiny1614 datasheet page 26

0x00 DEVICEID0 7:0 DEVICEID[7:0]
0x01 DEVICEID1 7:0 DEVICEID[7:0]
0x02 DEVICEID2 7:0 DEVICEID[7:0]
0x03 SERNUM0 7:0 SERNUM[7:0]
0x04 SERNUM1 7:0 SERNUM[7:0]
0x05 SERNUM2 7:0 SERNUM[7:0]
0x06 SERNUM3 7:0 SERNUM[7:0]
0x07 SERNUM4 7:0 SERNUM[7:0]
0x08 SERNUM5 7:0 SERNUM[7:0]
0x09 SERNUM6 7:0 SERNUM[7:0]
0x0A SERNUM7 7:0 SERNUM[7:0]
0x0B SERNUM8 7:0 SERNUM[7:0]
0x0C SERNUM9 7:0 SERNUM[7:0]

jashsheth, 2020-12-15:
The way to access it was "SIGROW.SERNUM"
I thought it to be direct like DDR registers in AVR
Thanks

closed 2020-12-15 by jashsheth

SpenceKonde, 2020-12-15:
That's because they're called `SIGROW.SERNUM0` and so on, (as members of the `SIGROW` struct), or equivalently, `SIGROW_SERNUM0` (as #defines). They've almost entirely done away with I/O registers that don't contain the name of the peripheral they are associated with. I can only think of 6 or 7 on these, depending on how you count (GPIOR0, GPIOR1, GPIOR2, GPIOR3, SREG, and SP / SPL & SPH - and in the Dx-series headers, they replaced the GPIOR ones with GPR.GPRn). It's really a huge improvement!

It's usually hard to guess what the #defines and struct members are actually called in the headers. I *always* end up with the device's io.h open when developing anything involving register access; if I can figure out the names of the headers, I'll need to check it for the name of a _bm/_gc/_gm/etc #define (I am trying vigorously to excise all the direct writing of numbers to peripheral registers from my cores... I used to be really bad about that, and I've realized it's worth the time to check names instead of writing numeric values to get the clarity and readability you get from using their defines for the bitfields. It's certainly more pleasant (if more verbose) than it was on the classic AVRs.

SpenceKonde, 2020-12-15:
By the way, on these parts, there aren't `DDR` registers ;-) It's `PORTx.DIR`, `PORTx.DIRSET/CLR/TGL` (for atomic access without having to turn off interrupts for a R/M/W cycle - unlike the `DDRx/INx/PORTx` registers on classic parts, the `PORTx` structs aren't in the low I/O space, nowhere near enough address space there), or `VPORTx.DIR` which is in the low I/O for atomic, single cycle, bit-flips via sbi/cbi, tests via sbis/sbic, and whole port writes via in/out (the compiler is smart enough to use them when it can (which means the bit and register must be compile time known for the "bit index" ones); naturally there's also `VPORTx.OUT`, `IN`, and `INTFLAGS`, and writing `VPORTx.IN` toggles the pin, unlike `PORTx.IN` (use PORTx.OUTTGL)