SpenceKonde/megaTinyCore #285: missing <Arduino>/hardware/megaTinyCore/megaavr/tools/prog.py

tgtakaoka, 2020-12-30:
I git cloned megaTinyCore.git under <Arduino>hardware on Linux and also downloaded and extracted "python3 package" as in "UPDI Programmer > From a USB-Serial adapter (pyupdi-style)" section of the README.md.
Then select Programmer > Serial Port and 4.7k (pyupdi style) on Arduino IDE.
The sketch (for ATtiny412) is compiled without error but can't upload via USB serial adapter because of the error in the title.

```
Sketch uses 298 bytes (7%) of program storage space. Maximum is 4096 bytes.
Global variables use 0 bytes (0%) of dynamic memory, leaving 256 bytes for local variables. Maximum is 256 bytes.
<Arduino>/hardware/megaTinyCore/megaavr/tools/python3/python3 <Arduino>/hardware/megaTinyCore/megaavr/tools/prog.py -t uart -u /dev/ttyACM0 -d attiny412 -v --fuses 2:0x02 6:0x00 8:0x00 -f/tmp/arduino_build_189876/cgen6309e.ino.hex -a write 
python3: can't open file '<Arduino>/hardware/megaTinyCore/megaavr/tools/prog.py': [Errno 2] No such file or directory
An error occurred while uploading the sketch
```

It seems the master branch is missing prog.py under megaavr/tools directory.

SpenceKonde, 2020-12-30:
> I git cloned megaTinyCore.git under hardware on Linux and also downloaded

Yeah, 2.2.0 isnt out yet and the new took isnt in there yet

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2021-01-06:
2.2.0 is being released in minutes; the files are in now.

closed 2021-01-06 by SpenceKonde