SpenceKonde/megaTinyCore #286: Wire.endTransmission() hangs after version upgrade

vitadostal, 2020-12-30:
Hi,
I am using only <SoftwareSerial.h> and <Wire.h> libraries.
After upgrading megaTinyCore to version 2.1.5 from 1.1.10, Wire.endTransmission(); hangs. Even an ordinary I2C scanner hangs on the same command.
Chip: ATtiny 3216 1Mhz.
Am I doing something wrong? I don't think it is an HW issues because after downgrade to 1.1.10 everything works as expected.
Vita

SpenceKonde, 2020-12-30:
Do you have external pullup resistors in place? (if using a breakout board, verify that they are actually connected - some require connecting a solder bridge in order to connect their pullup resistors). 4.7k is a good value for external pullup resistors, barring other design requirements.

Older versions turned on the internal pullups; it was discovered that this interfered with some planned enhancements during 2.1.x development, and, since the internal pullups are about an order of magnitude weaker than the I2C spec requires anyway, it was decided that they would no longer be enabled moving forward, on the grounds that it's better for I2C to immediately fail when the required external pullups are not present, than to have it work under simple/minimal configurations, and then fail unpredictably as more devices or longer wires are connected.

vitadostal, 2020-12-30:
I decided not to use these resistors on the new PCB (own design with minimal distance between chip and one I2C device). `pinMode(i2cPins, INPUT_PULLUP);` always worked fine for me with several generations of ATtiny chips. So if I get you right I need them for 2.x.x versions (not easy to test on finalized very small PCB). Thanks for your fast reply.

vitadostal, 2020-12-31:
With two 10K or 4K7 resistors 2.1.5 works as expected.

closed 2020-12-31 by vitadostal