SpenceKonde/megaTinyCore #287: ArduinoIDE Board Manager error

Targett363, 2021-01-02:
Can't get the Board Manager to find the megaTinyCore.
Hoping it's just a typo in the .json.
IDE shows this error message.

```
Skipping contributed index file C:\Users\targe\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('<' (code 60)): expected a valid value (number, String, array, object, 'true', 'false' or 'null') at [Source: (FileInputStream); line: 2, column: 2]
```

SpenceKonde, 2021-01-03:
I cannot reproduce this issue. 
Can you access http://drazzy.com/package_drazzy.com_index.json? 

Is C:\Users\targe\AppData\Local\Arduino15\package_drazzy.com_index.json valid json? It could bre locally caching an http error maybe?

Targett363, 2021-01-03:
Looks like it’s Sky’s fault!
I’ll be home Monday and using a better internet provider!
[image]

Thanks so much!

closed 2021-01-03 by Targett363

SpenceKonde, 2021-01-03:
Huh - really wonder why Sky is blocking me :-P

Does it just block everything that's not https, by default?

(note that we will move to https when I finally have time to get azduino.com sorted out)