SpenceKonde/megaTinyCore #288: flashing ATtiny412 via UDPI fails

jstraub, 2021-01-04:
Hi

I am trying to program the ATtiny412 I bought from you (https://www.tindie.com/products/drazzy/attiny412402-dev-board-arduino-compatible/) a while ago and I am failing. I am using a Arduino Pro Mini with jtag2udpi flashed onto it (like https://github.com/SpenceKonde/jtag2updi). I did install the https://github.com/SpenceKonde/megaTinyCore and trying to just flash a blink arduino program.

I got the UDPI version of the ATtiny412 from you. It seems the chip has a blink program already flashed onto it? What is the default fuse configuration? I got the 3.3V version. Should I have to flash the fuses?

I just end up getting the following if I am trying to upload a blink program (I changed the on of blink time to be able to distinguish from the default one and it is definitely not being flashed):
```
Sketch uses 738 bytes (18%) of program storage space. Maximum is 4096 bytes.
Global variables use 10 bytes (3%) of dynamic memory, leaving 246 bytes for local variables. Maximum is 256 bytes.
avrdude: jtagmkII_recv(): msglen 8912897 exceeds max message size 100000, ignoring message
avrdude: jtagmkII_recv(): msglen 8912897 exceeds max message size 100000, ignoring message
avrdude: jtagmkII_recv(): msglen 8912897 exceeds max message size 100000, ignoring message
avrdude: jtagmkII_recv(): msglen 8912897 exceeds max message size 100000, ignoring message
avrdude: jtagmkII_set_devdescr(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
```

Any idea whats going wrong?

SpenceKonde, 2021-01-04:
Hmmm, that's a very strange one - it's clearly a problem in communicating with jtag2updi, but I can't say I've ever seen that error before...

Does connecting a ~10uF (exact value doesn't matter - anything more than a few uF will work) cap between Reset and Ground on the Pro Mini (or disconnecting the DTR line - whichever is easier) - I'm wondering if Autoreset combined with the pro mini bootloader is causing the problem, resulting in avrdude talking to the pro mini's bootloader instead of jtag2updi...

Also try a manual reset of the pro mini. 

Let me know if this does or doesn't work

jstraub, 2021-01-04:
I have tried this with and without the DTR line connected to the Pro Mini -
same result. I wish I had known about the capacitor trick :) now I have one
Pro mini without auto reset...

I have also tried to hit the reset button. I was not quite sure when to hit
it though? Right before trying to program? Do I have to press it and then
start programming and then release it?

SpenceKonde, 2021-01-04:
With autoreset disabled and a manual reset performed at least once after power was applied, you should be good on that front. 

That's a wacky error, I'm... not immediatelyu sure where to go on this....

One thing that son your message doesnt quite add up (this can be either a red flag or a red herring) - you say you have a Pro Mini that doesn't have autoreset. But how could that be, when all that is required is disconnecting a wire between the serial adapter and the pro mini?

jstraub, 2021-01-04:
I clipped the DTR pin on the serial connection of one of my pro minis. That way there should be no more auto reset, right? I wish I had known about the capacitor trick and spared my board that way ;).

One question I was wondering on: is there a way to verify that jtag2updi is installed correctly on the arduino pro mini? Will the LED flash if UPDI programming is happening for example?

jstraub, 2021-01-04:
Here is a picture of the setup in case you see something wrong?

jstraub, 2021-01-04:
Here is the setup I am uisng (note the clipped pin on the pro mini serial connector):
[image]
See anything wrong?

jstraub, 2021-01-05:
I just read that the defaults for jtag2updi could be a problem on lower clock rates. 

My mini pro is 8Mhz and the ATtiny412 is also 8Mhz. Could that be the problem?

jstraub, 2021-01-05:
So I tried reducing the BAUD rate to 80k but that also did not help. 
Then I tried to measure the signal on the D6 pin of the Arduino Pro Mini and I am not getting any signal when I try to program.
So something is off with the jtag2udpi. Any thoughts on how to debug?

SpenceKonde, 2021-01-05:
I actually got like, halfway into wiring up a jtag2updi with a pro mini, and then got distracted...

Say, you're not, by any chance, using a version of the jtag2updi library that you pulled more than like, a month and a half ago are you? That version was totally hosed.

SpenceKonde, 2021-01-05:
( realized that I'd never actually made one with a Pro Mini, always with a nano... but I do say it will work, and it goddamned well should...

pfoun, 2021-01-05:
The pro mini works well.  Uploaded firmware to the pro mini from here:
https://github.com/ElTangas/jtag2updi
using USBtinyISP programmer.  No bootloader on the pro mini.  Resistor on UPDI is 1k.
Uploaded successfully to ATtiny1614 on a generic breakout board.  Here a photos of the settup:
[image]
[image]

jstraub, 2021-01-05:
@SpenceKonde I did clone the jtag2udpi library only a day or so ago. 
I found a old Arduino Nano and was able to get it to work and flash the ATtiny412. 

Thanks @pfoun for trying it out with the mini pro! The difference is that you had no bootloader on your mini pro and I did. So seems pretty clear that the mini pro bootloader is the problem. Does anybody know how to remove a boot loader? ^^

pfoun, 2021-01-05:
Program the jtag2updi firmware into the pro mini with a ICSP programmer, such as AVR ISP, USBtinyISP, USBasp, and the bootloader will be inactive.  If you do not have a ICSP programmer, you can make one from a Arduino.  Search for Arduino as ISP, you will find instructions online.

SpenceKonde, 2021-01-06:
Added mention in the section on building jtag2updi programmer - though note that we now support programming with just serial adapter and resistor like *pyupdi*

closed 2021-01-06 by SpenceKonde

jstraub, 2021-01-06:
Cool thanks guys for helping debug. :)