SpenceKonde/megaTinyCore #289: Issue with uploading sketches using Optiboot

freemovers, 2021-01-08:
I have used Optiboot a couple of times about 6 months ago, but today I got an issue with uploading sketches. I use PA0 as reset pin and that seems to work. AVRDUDE even gets the device signature, but throws the following error:

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\svandebor\Documents\Arduino\hardware\megaTinyCore\megaavr/avrdude.conf"

         Using Port                    : COM11
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATtiny1616
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 9.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

An error occurred while uploading the sketch
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9421 (probably t1616)
avrdude: reading input file "0x02"
avrdude: writing fuse2 (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.00s

avrdude: 1 bytes of fuse2 written
avrdude: verifying fuse2 memory against 0x02:
avrdude: load data fuse2 data from input file 0x02:
avrdude: input file 0x02 contains 1 bytes
avrdude: reading on-chip fuse2 data:

Reading | avr_read(): error reading address 0x0000
    read operation not supported for memory "fuse2"
avrdude: failed to read all of fuse2 memory, rc=-2

avrdude done.  Thank you.

```

SpenceKonde, 2021-01-08:
Yikes, thanks for reporting that. It's a new bug introduced in 2.2.0 - as you can see, it's attempting to write the fuses - because the recipes in platform.txt are not correct for optiboot board definitions.

SpenceKonde, 2021-01-08:
Can you confirm that this adjustment to platform.txt fixes it? I'd like another set of eyes on it before I do a bugfix release, but I need to get that fix out ASAP - it's a pretty bad bug!

closed 2021-01-08 by SpenceKonde

SpenceKonde, 2021-01-08:
I satisfied myself testing this, 2.2.1 is released, it fixes this.

freemovers, 2021-01-09:
Sorry, I was not able to test earlier, but it is working great! Thank you so much.

commit a24af0c4 (2021-01-08): Fixes #289 - optiboot uploads fail, non-optiboot doesn't set fuses

2.2.0 introduced a new feature of writing "safe" fuses with every upload when a programmer was in use. Unfortunately, I got it backwards. So it attempted to set the fuses when programming through optiboot (which failed, obviously), whilest not setting the fuses on normal uploads. This is extra critical because of how loudly I have been trumpeting the fact that you don;t need to burn bootloader to change  between 20 and 16 MHz derived speeds.