SpenceKonde/megaTinyCore #290: 2.2.1 broken?

iansexton, 2021-01-08:
Fails to load with boards manager:

Index error: could not find referenced tool name=python3 version=3.7.2-post1 packager=megaTinyCore_pmptest

SpenceKonde, 2021-01-08:
Argh! forgot to change the packager when I moved it out of the test package and into the real one, but it didn't show up in my tests because it recognized the name from the other package... x_x

Should be sorted out now.

iansexton, 2021-01-08:
Nope

 Error downloading https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1/megaTinyCore-2.2.1.tar.gz
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1/megaTinyCore-2.2.1.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1/megaTinyCore-2.2.1.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more

per1234, 2021-01-08:
According to the release notes:
https://github.com/SpenceKonde/megaTinyCore/releases/tag/2.2.1-manual
>Note that we are now only doing one "release" per version.
>
>The link that GitHub calls "source code" is the manual installation version (megaTinyCore-2.2.1-manual.tar.gz), while megaTinyCore-2.2.1.tar.gz, attached, is the board manager version.

It looks like the archive URL should be:
https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1-manual/megaTinyCore-2.2.1.tar.gz
rather than the current:
https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1/megaTinyCore-2.2.1.tar.gz

SpenceKonde, 2021-01-08:
Yup, i just fixed it, it should be working now. x_x

closed 2021-01-09 by SpenceKonde