SpenceKonde/megaTinyCore #291: Can't find create_disassembler_listing.sh in Linux Mint 20

pfoun, 2021-01-09:
I can't see why I'm getting this error in Linux Mint 20:

Arduino: 1.8.13 (Linux), Board: "ATtiny1614/1604/814/804/414/404/214/204, No (saves space), ATtiny1604, 20 MHz Internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (pick this unless you have an HV UPDI programmer!), 8ms, Closer to 5v"

fork/exec /home/owner/.arduino15/packages/megaTinyCore/hardware/megaavr/2.2.1/create_disassembler_listing.sh: no such file or directory
Error compiling for board ATtiny1614/1604/814/804/414/404/214/204.

The file is there and it is executable.  Version 2.2.0 and 2.2.1 both give this error.  Previous versions did not.  The error does not occur in windows.

SpenceKonde, 2021-01-09:
Anyone got any ideas here? 

I haven't changed anything related to this, as far as I know...  I don't have a linux system handy to investigate with.

per1234, 2021-01-09:
I am able to reproduce the same error.

When I tried running the script from the command line, I got a more useful error message:
```
bash: /home/per/.arduino15/packages/megaTinyCore/hardware/megaavr/2.2.1/create_disassembler_listing.sh: /bin/bash^M: bad interpreter: No such file or directory
```
`^M` is the carriage return character. Once I changed the script to use LF line endings the error when compiling went away.

I checked and found the line endings in the script here in the repository (https://github.com/SpenceKonde/megaTinyCore/blob/2.2.1-manual/megaavr/create_disassembler_listing.sh) are the Linux-style LF, but line endings of the script in the release archive (https://github.com/SpenceKonde/megaTinyCore/releases/download/2.2.1-manual/megaTinyCore-2.2.1.tar.gz) file are the Windows-style CRLF.

So I guess somehow during the creation of the Boards Manager release archive the line endings were changed from LF to CRLF.

SpenceKonde, 2021-01-09:
Ugh........ 

No, it looks like github is going line-ending correction, but that doesn't happen when I create the archive on my local machine (which is windows)

Thanks for the investigation. I won't be able to look at this until later tonight.

per1234, 2021-01-09:
Perhaps it's your Git line ending configuration:
https://git-scm.com/docs/git-config#Documentation/git-config.txt-coreautocrlf
This setting will cause automatic conversion from LF to CRLF on checkout if set to `true`.

I prefer to set it to `input` for my own repositories because I always want LF line endings, and `false` when contributing to other people's repos to maintain whichever line endings they already have in place (unless my commit is specifically intended to change the line endings).

pfoun, 2021-01-09:
@per1234 Thank you.  I edited the script and now the sketch compiles.

SpenceKonde, 2021-01-11:
I think this should be sorted out with 2.2.3

closed 2021-01-11 by SpenceKonde