SpenceKonde/megaTinyCore #292: Attiny 3217 Dont work with I2C

kelroy1990, 2021-01-10:
Hi, at this moment i am trying to play and understand this Tinny core world, thanks Spence to allow it to work with arduino.

Currently now i am testing it on the I2C bus, but a strange behaivour happends, when i start a Wire Transmission the CPU get stucked.

To understand that it get stucks i am using a led blinky as debug.

The board i am using its a Curiosity Nano, as it was one i have already buyed. 

Any help will be appreciate, as i really dont understand why its it happening. Does somebody test it on the 3217?¿.

```{c}

#include <Wire.h>
 
 
void setup()
{
  Wire.swap(0);
  Wire.begin();
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("I2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");
 
  delay(5000);           // wait 5 seconds for next scan
} 
```

Regards, Carlos.

SpenceKonde, 2021-01-11:
Did you include the required pullup resistors as described in the readme?

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-01-11:
```
Warning: I2C requires external pullup resistors
Earlier versions of megaTinyCore enabled the internal pullup resistors on the I2C pins. This is no longer done; they are miles away from being strong enough to meet the I2C specifications - it was decided that it is preferably for it to fail consistently without external ones than to work under simple conditions with the internal ones, yet fail under more demanding ones (more devices, longer wires, etc).
```

kelroy1990, 2021-01-11:
Hi Spence, the board from microchip i have on the lab have the Pull-up resistors solder in.

To go further i program the board with the AVR framework and microchip studio, it take some hours but at last i made the i2c working, so the problem isnt the board, maybe some issue on the core code?.

Did you code the i2c on interrupt or polled mode?, on the interrupt mode we should take care about the PORT.CONTROL registers, i am going to read the core with more detail to see this I2C part.

SpenceKonde, 2021-01-12:
Other people have reported that I2C works. Do verify that the pullups are actually connected (ie, measure resistance from the pin to Vcc) and don't need a solder-bridge or jumper to enable them... as I recall (though I don't have any to play with!), Microchip's boards don't have the pullups connected by default, in case you want to use the pins for other purposes.

I didn't code _anything_ regarding I2C :-P . Except for a few changes (see history of the file) to support new features (all of which are either PRs or were made after others tested the changes), it's the same code the official Uno WiFi Rev. 2 and Nano Every use, They have the same I2C peripheral. Personally, I've never even wired up I2C on a modern AVR - have been too busy maintaining the cores to work on any of my projects which will make use of I2C if I ever have a chance to work on them again.

per1234, 2021-01-12:
I don't know whether it's relevant, but it looks like you haven't pulled in this fix from upstream: https://github.com/arduino/ArduinoCore-megaavr/pull/90

kelroy1990, 2021-01-12:
After review the board it have a 4,7K pull-up solder, so its well done.

This is how i activate the I2C registers on the atmel studio side:
 **Interrupt mode, if you want polled mode you should put TWI_RIEN_bp to 0 and TWI_WIEN_bp to 0**

```{c}
#define TWI0_BAUD(F_SCL, T_RISE) ((((((float)4000000 / (float)F_SCL)) - 10 - ((float)4000000 * T_RISE / 1000000))) / 2)

	TWI0.MBAUD = (uint8_t)TWI0_BAUD(100000, 0); /* set MBAUD register */

	TWI0.MCTRLA = 1 << TWI_ENABLE_bp        /* Enable TWI Master: enabled */
	              | 0 << TWI_QCEN_bp        /* Quick Command Enable: disabled */
	              | 1 << TWI_RIEN_bp        /* Read Interrupt Enable: enabled */
	              | 0 << TWI_SMEN_bp        /* Smart Mode Enable: disabled */
	              | TWI_TIMEOUT_DISABLED_gc /* Bus Timeout Disabled */
	              | 1 << TWI_WIEN_bp;       /* Write Interrupt Enable: enabled */

```

As i have some time i gonna see under the oscilloscope and under the saleae the i2C output to try to understand the problem.

SpenceKonde, 2021-01-17:
Thanks, I just brought in that change - let me know if that improves things.

SpenceKonde, 2021-01-19:
FYI, I bungled the new changed. I2C is... quite possible completely hosed ATM.

SpenceKonde, 2021-01-23:
I'm suspicious that the issue may in fact have been related to some tweaks to digital I/O - I don't remember exactly what changes were made and when, but if they caused the PORTx.OIT register to be set to 1 for either of the I2C bits, well, that'd do it.... because basically every AVRxt device released has an errata with the TWI interface, where it overrides DIR but not OUT...

SpenceKonde, 2021-01-23:
But anyway, I hardened wire against that -let me know if you see different vehavior with it. Have a lot on my plate so haven;t done much testing on it.

kelroy1990, 2021-02-04:
Hi Spence, the i2c on the last update seems to work again. At least with the LIS3DR that i am testing.

SpenceKonde, 2021-02-04:
Hurray!

closed 2021-02-04 by SpenceKonde