SpenceKonde/megaTinyCore #294: New UDPI Serial upload

chipperdoodles, 2021-01-12:
Hi! this isn't an issue more of a clarification question, but wasn't sure if there was a better place to post this!

I'm designing a board with the intention of using this core and was excited to see the new option of serial udpi programming. I just wanted to make sure I read the instructions right:

[image]

Is this the intended way to wire up the resistors for the serial -> udpi uploader? (the usb-uart chip will be on the same board as the 3217 so the values are based on that and not using external serial breakout board).

Thank you so much! I like these new lil avr and all your work on the core has enabled me to mess around with them!

SpenceKonde, 2021-01-12:
Lose the 470 ohm resistor if you arent going to hook up other programmers -
it's only there to guarantee that an external programmer, even one without
the series resistor, can program it safely. While it doesnt interfere with
programming via just a serial adapter, it's entirely unnecessary.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

chipperdoodles, 2021-01-12:
Thank you!  that would help save some space. Really appreciate the response!

closed 2021-01-12 by SpenceKonde