SpenceKonde/megaTinyCore #295: Attiny402 + WS2812 Led Strip

Wallteri, 2021-01-12:
Hello, i wanted to try atiny402 with ws2812 Led strip. So I used the example from tinyNeoPixel library. Everything works fine until I used more than 25 LEDS(PIXELS), then light only first LED(PIXEL) other LEDS are off. Is there any limitation with number of LEDS(PIXELS) or what can be wrong?
Thanks for answer.

Code:
`
#include <tinyNeoPixel.h>

// Which pin on the Arduino is connected to the NeoPixels?
#define PIN            2

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS      10

// When we setup the NeoPixel library, we tell it how many pixels, and which pin to use to send signals.
// Note that for older NeoPixel strips you might need to change the third parameter--see the strandtest
// example for more information on possible values.
tinyNeoPixel pixels = tinyNeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

int delayval = 100; // delay for half a second

void setup() {

  pixels.begin(); // This initializes the NeoPixel library.
}

void loop() {

  // For a set of NeoPixels the first NeoPixel is 0, second is 1, all the way up to the count of pixels minus one.

  for (int i = 0; i < NUMPIXELS; i++) {

    // pixels.Color takes RGB values, from 0,0,0 up to 255,255,255
    pixels.setPixelColor(i, pixels.Color(150, 0, 0)); // Moderately bright green color.

    pixels.show(); // This sends the updated pixel color to the hardware.

    delay(delayval); // Delay for a period of time (in milliseconds).

  }
}`

SpenceKonde, 2021-01-13:
My guess is you don't have enough RAM. 

I would try using tinyNeoPixel_static (note that you also need to adjust the size of the pixels buffer to change the length of the string) - that way it will tell you how much memory you're using at compiletime.

SpenceKonde, 2021-01-13:
Closing issue, as - without more evidence - there's no reason to suspect it is a defect in the core or included libraries. I will still see and respond to messages here, just want to keep the issue list populated with actual action items relating to the core and included libraries..

closed 2021-01-13 by SpenceKonde

SpenceKonde, 2021-01-13:
In general, I would not recommend the "normal" (ie, not "static") version of the library - you'll also notice a significant flash savings by going to the static version. "static" refers to the fact that the memory for the frame buffer is statically allocated instead of dynamically allocated as in the "stock" neopixel library. At the cost of not being able to change the length of the strip at runtime (I don't know about you, but I rarely find myself dealing with a strip of neopixels that changes length while the code is running >.> ), you can ditch all the baggage associated with malloc:
* malloc is bulky, particularly relative to the flash on the 4k parts.
* dynamically allocated memory is not counted in the memory usage shown when a sketch is compiled. This particularly sucks when you're working with something with such a small amount of memory).
* if you are frequently allocating and dealocating memory, you can fragment the heap (a great way to trigger this is to use the arduino String (big S) class, and append characters to it one-at-a-time. Which is something people do all the time) - embedded systems don't have any mechanism to defragment the free memory - when there's not enough to allocate the required memory, malloc just fails (and basically no arduino code detects or handles this case - you can't really spare the overhead for that anyway). This is why the experienced Arduino people tell newbies to avoid String (which in turn is why everyone tells folks not to use them on the forums; most of the people repeating that warning don't know why they're bad, but they know everyone told them to avoid them)