SpenceKonde/megaTinyCore #296: java NullPointerException on board update/install

mart-jakobson, 2021-01-13:
Was just about to start moaning about Arduino again, but noticed all other boards I have installed, seem to work.

Darned Win10.
Arduino 1.18.3
Installed megaTinyCore 2.2.0

Repro: Try to update, downgrade or remove megaTinyCore.
No repro: Try same things to ATTinyCore 1.4.1 or DxCore 1.2.0

Tried same on Linux, no problems there (Arduino 1.18.3) Had mTC 2.1.5 installed, upgraded to 2.2.0, then upgraded to 2.2.3 no probs.

```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:276)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)
	... 1 more

```

```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:276)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:210)
	... 1 more
```

SpenceKonde, 2021-01-13:
It doesn't reproduce for me; it's a system problem. I don't recall how to fix that one.

mart-jakobson, 2021-01-13:
Removing core manually and reinstalling thru IDE seems to have fixed everything.

closed 2021-01-13 by mart-jakobson

SpenceKonde, 2021-01-13:
Thanks for letting me know, I'll try to remember that fix if someone else runs into it.