SpenceKonde/megaTinyCore #298: Prescaler & PIT-Interrupt Issue

DerPicknicker, 2021-01-15:
Hey, 

I try to use your example Code for PowerSaving (Using SleepMode on the Attiny) I found an Issue and I cannot fix it. If I use the PIT-Interrupt AND the Prescaler the LED WONT blink. But if I use ONLY the PIT-Interrupt the LED goes on and off ... I hope you can give me some Information about it. 

Have a nice Weekend. 

My Code: 
```
#include <avr/sleep.h>

void RTC_init(void)
{
   while (RTC.STATUS > 0)
   {
      ;
   }

  RTC.CLKSEL = 0x0;
  RTC.CTRLA = (0x2<<3) | 0x1;
  RTC.PITINTCTRL = 0x1;
  RTC.PITCTRLA = (0xd<<3) | 0x1;
}

ISR(RTC_PIT_vect)
{
  RTC.PITINTFLAGS = 0x1;
}

void setup() {
  RTC_init();
  pinMode(6, OUTPUT);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
}

void loop() {
  sleep_cpu();
  digitalWrite(6, CHANGE);
}
```

SpenceKonde, 2021-01-15:
Off the top of my head - no compile test even, don't know what part you're using, etc - so adjust as needed. 

I have no f'ing clue, quite frankly, other than the fact that the synchronization thing is weird, and the sort of thing that if you don't check your self could behave in a way other than what you are expecting (basically everything that has wacky sync behavior, when I first start working with the peripheral, I never take the datasheets word for what it's doing. 

Also... why don't you use the #defines from the headers? It is so much easier to go back and read your code in the future!

```
byte record[6];
byte pits[6];

void RTC_init(void)
{
  while (RTC.STATUS > 0 || RTC.PITSTATUS)
  {
    if(record[0]==0) { // first pass, when it's probably the same as when we one line prior, we record what we saw... 
      record[0]=RTC.STATUS | (RTC.PITSTATUS ? 0x80 : 0 );
    }
  }
  RTC.CLKSEL = 0x0; //uh... you know you don't actually need this, power on reset state is 0...
  //.... but you should run it with this data gathering at least once so you'll know if this.
  // may have been involved.  Maybe changing the clock source triggers one of the synch cycles?
  while (RTC.STATUS > 0 || RTC.PITSTATUS)
  {
    if(record[1]==0) {
      record[1]=RTC.STATUS | (RTC.PITSTATUS ? 0x80 : 0 );
    }
  }
  // Trust nobody - record those status bits every time we touch the regs, because we think it's some stupid sync problem!
  RTC.CTRLA = RTC_PRESCALER_DIV4_gc | RTC_RTCEN_bm;

  while (RTC.STATUS > 0 || RTC.PITSTATUS)
  {
    if(record[2]==0) {
      record[2]=RTC.STATUS | (RTC.PITSTATUS ? 0x80 : 0 );
    }
  }

  RTC.PITINTCTRL = RTC_PI_bm;
  
  while (RTC.STATUS > 0 || RTC.PITSTATUS)
  {
    if(record[3]==0) {
      record[3]=RTC.STATUS | (RTC.PITSTATUS ? 0x80 : 0 );
    }
  }

  RTC.PITCTRLA = (RTC_PERIOD_CYC16384_gc | RTC_PITEN_bm);
  
  while (RTC.STATUS > 0 || RTC.PITSTATUS)
  {
    if(record[4]==0) {
      record[4]=RTC.STATUS | (RTC.PITSTATUS ? 0x80 : 0 );
    }
  }
}

ISR(RTC_PIT_vect)
{
  // assuming we used PA5.
  // this will toggle another LED in another way - and also (by being on at 50% brightness constantly) tell you if it's getting stuck in the interrupt somehow.

  VPORTA.IN |= 1<<5; 
  RTC.PITINTFLAGS = RTC_PI_bm;
}

void setup() {
  pinMode(6, OUTPUT);
  // another pin to put another LED on....
  // I wanted it set output BEFORE the interrupt gets enabled, though
  // that way, if some interrupt horror story is going on and it's
  // not even making it out of RTC_init() before it lands in eternal 
  // interrupts, you will be keenly aware.
  pinMode(PIN_PA5, OUTPUT); 

  Serial.begin(115200);
  //assuming it's not running at a super low clock speed, this is 2021, it's time to get out of the habit of reflexively using 9600 baud. 
  RTC_init();
  #if (MEGATINYCORE_NUM > 0x02010401) // has printHex()
  Serial.printHex(record,5,' ');
  Serial.println();
  Serial.println("Now registers are: ");
  Serial.printHex(RTC.CTRLA);
  Serial.print(' ');
  Serial.printHex(RTC.CLKSEL);
  Serial.print(' ');
  Serial.printHex(RTC.PITCTRLA);
  Serial.print(' ');
  Serial.printHex(RTC.PITINTCTRL);
  Serial.println();
#else //lame-serial-print
  Serial.print(record[0],HEX);
  Serial.print(' ');
  Serial.print(record[1],HEX);
  Serial.print(' ');
  Serial.print(record[2],HEX);
  Serial.print(' ');
  Serial.print(record[3],HEX);
  Serial.print(' ');
  Serial.print(record[4],HEX);
  Serial.println();
  Serial.println("Now registers are: ");
  Serial.print(RTC.CTRLA,HEX);
  Serial.print(' ');
  Serial.print(RTC.CLKSEL,HEX);
  Serial.print(' ');
  Serial.print(RTC.PITCTRLA,HEX);
  Serial.print(' ');
  Serial.println(RTC.PITINTCTRL,HEX);
#endif
  Serial.println("hopefully-not-eternal sleep")
  Serial.flush(); // wait for serial to finish printing that.
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
}

void loop() {
  sleep_cpu();
  digitalWrite(6, CHANGE);
}
```

freemovers, 2021-01-16:
Look like you can not make changes to the RTCdirectly after the PIT is enabled. I was able to resolve the issue by changing initializing the RTC as follows:
```
void RTC_init(void)
{
  while (RTC.STATUS > 0 || RTC.PITSTATUS);
  RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;
  RTC.PITINTCTRL = RTC_PI_bm;
  RTC.PITCTRLA = RTC_PERIOD_CYC16384_gc | RTC_PITEN_bm;  
  while (RTC.STATUS > 0 || RTC.PITSTATUS);
  RTC.CTRLA = RTC_PRESCALER_DIV4_gc | RTC_RTCEN_bm
}
```

SpenceKonde, 2021-01-17:
@DerPicknicker  - did you have any luck with this? Not sure how to interpret the emojis 

There's definitely a bit of a learning curve with all this async stuff now (but some of it is oh so powerful)... closing (though I still see notifications) since it sounds like the issue has been solved or understood now, and in any event there's no defect in the core involved here

closed 2021-01-17 by SpenceKonde

DerPicknicker, 2021-01-18:
Sorry for my late Reply.. I will need some time to adjust your code. I can post the serial output soon.

I thought it was an issue of the core. Like an wrong Register or something like that. 

The code from freemovers solved the issue. It is an sync error.