SpenceKonde/megaTinyCore #300: digitalWrite disables wrong WO pin

fehrudi, 2021-01-19:
With 1616/816 I tested the TCA0 example 2 - PWM on WO0/PB0, which worked fine until I wanted to use PA3 as output.
The turnOffPWM() function in wiring_digital.c disables the wrong PWM pin, there is no check for single/split mode.
In single/normal mode the WO5 (PA3) can't be active, the TCA0.SINGLE.CTRLB CMPnEN0 (PB0) is at same address/bit as TCA0.SPLIT.CTRLB HCMP0EN.

> pinMode(PIN_PB0, OUTPUT); 
TCA0.SPLIT.CTRLA=0; 
TCA0.SPLIT.CTRLESET=TCA_SPLIT_CMD_RESET_gc|0x03; 
  TCA0.SPLIT.CTRLD=0; 
  TCA0.SINGLE.CTRLB=(TCA_SINGLE_CMP0EN_bm|TCA_SINGLE_WGMODE_SINGLESLOPE_gc); //Single slope PWM mode, PWM on WO0
  TCA0.SINGLE.PER=Period;  
  TCA0.SINGLE.CMP0=0;
  TCA0.SINGLE.CTRLA=TCA_SINGLE_ENABLE_bm; 
setFrequency(frequency);
  setDutyCycle(64);
pinMode(PIN_PA3, OUTPUT);
digitalWrite(PIN_PA3, HIGH); //At this line, the timer on PB0 is disabled

SpenceKonde, 2021-01-19:
Well, I do warn about this sort of thing... 

```
Warning: Direct Register Manipulation
If you are manually manipulating registers controlling a peripheral, you should not count on the behavior of API functions that relate to the same peripheral, nor should you assume that calling said API functions will not adversely impact the rest of your application. For example, if you "take over" TCA0, you should not expect that using analogWrite() - except on the two pins on the 20/24-pin parts controlled by TCD0 - will work for generating PWM; indeed you should expect that it will break whatever you are doing with TCA0.
```

However - there is certainly *one* workaround you can use right now - 2.2.x adds the digitalWriteFast() calls, As long as the pin number is constant and compiletime known, digitalWriteFast(pin,HIGH), digitalWriteFast(pin,LOW);, and digitalWriteFast(pin,CHANGE);  optimize down to a very small, very fast operation (Don't recall what it optimizes down to if the code is deciding the state at runtime (eg, `digitalWriteFast(PIN_PB0,(some_condition?HIGH:LOW))` )but if you do, it turns into an SBI or CBI instruction - runs in 1 clock cycle instead of... however damned many digitalWrite normally takes.

But among the limitations of digitalWriteFast(), it doesn't turn PWM off :-P

The real problem is that, when one starts trying to cover anything beyond the default timer configuration the core provides you with, the overhead all those tests and setting bits when the tests connect... it becomes prohibitive in terms of flash usage (remember, some people are using 4k parts - only doing basic stuff, it's a basic chip, but whether or not they use those features, they have to pay the cost of testing for the state of things like that) . It's not hard to blow through 10% of their available flash after the overhead for the absolute basics! You would be amazed how many people are using the smaller parts (no, I don't get it either it's not like the price savings is huge or anything - but they do exist, and routinely complain about how there isn't enough flash for basic stuff... on their parts...).

Really, I think your specific problem is an example of the fact that the digital I/O functions are terrible, but there's not really anything between those writing registers directly. One really wants a more "powerful" version of digitalWrite/Read, and pinMode... It's something I'm thinking about... but not sure the right approach - because of the huge amount of code written for the classic wiring I/O functions (that's where digitalRead and it's ilk came from). If Im being honest, I very often just use the port writes myself - but then you can't have nice defined like LED_PIN and have them be usable in the same way. 

Another problem with *improving* these things too much is that people have written code that *unbeknownst to them, generally) onbly works as long as digital I/O isn't lightning fast; that's why we don't have automatic-fast (or, in practice, moderately-faster-and-smaller-I/O) for example... While I would certainly be happy to make everyones pin ready and writes less miserably slow, I wouldn't call it a win if I make 80% of tyhem really fast, but the rest stay slow, so some pins are "fast" and others are "slow" - that kind of thing is like... pretty lame, you know? But yeah, I feel your pain here, but I don't have a good answer other than not using the Arduino API functions...
 
One of the other things I have been pondering (though it works much better for the AVR Dx-series, since they have more formulaic peripheral layouts and more flash to fit this sort of bloat) is ways to provide an Arduinoy wrapper around PWM pin-swaps (for example, make it work more like the ATTin167).... but I have so many exciting things I want to do with the cores :-( and so little time to do them....

SpenceKonde, 2021-01-27:
I just got an idea.... What about a takeOverTCA0 function? if you called it it would stomp out one bit of a byte checked every time analogWrite or turnOffPWM checks what timer a pin has... I already have a few other "reach into the core" type functions (for example, the ones to manipulate millis timekeeping)so there is absolutely precedent for throwing weirdo non-standard function calls into the core.... this would be slightly more public-facing, but...

One would get 8 peripherals to take over from one byte, and the cost of checking that is only 6 bytes each for turnOffPWM and analogWrite (I'd have the byte be inverted; for the timers - the most access-heavy ones out of the lot) I could then bitwise AND it with the result we got back from digitalPinToTimer...?... 

BUT - the real gem in this approach is that if the compiler saw that you never touched the variable? *it would optimize it away* I almost didn't realize that at first, because I'm always more concerned about getting rid of optimization I don't want than the other way around...

SpenceKonde, 2021-01-27:
Would naturally also come with a takeOverTCD0 (man - I just wish I could know when someone was using it - I'd get a twinge of schadenfreude out of that - because I've used that balky async monstrosity myself to implement analogWrite - and I'd just imagine them..... but, I set the compare registers! I can read them back! Why is it ignoring them?? I told it to update at end of cycle, why isn't it updating? I read this chapter of the datasheet ten times and it still might as well have been written in greek! Aaaaah... yeah, TCD is an awful peripheral to work with! Powerful (didja know you can squeeze an extra channel out of TCD0 and CCL LUT0? (any even numbered LUT works)).... but nuclear technology is powerful too, but that doesn't mean one would recommend taking a stroll through a nuclear reactor o

SpenceKonde, 2021-02-10:
takeOverTCA0 is in 2.2.7 - you should be able to call that, and you should expect that: 
analogWrite() will treat the pins like normal digital pins. 
digitalWrite() will not turn off PWM on TCA0-controlled PWM pins. 
TCA0 will be disabled
If we're lucky, TCA0 will have been reset with the hard reset command (but check the registers to be sure it worked - I have seen weird behavior with the hard reset command) - if it works like it did in test, you'll have TCA in power on configuration, not even split mode! But that last point I won't guarantee :-P the other three should definitely do that.

closed 2021-02-10 by SpenceKonde