SpenceKonde/megaTinyCore #301: Error downloading some tools during boards manager installation.

sideswiper22, 2021-01-22:
Just encountered this issue today after trying to update the core to the latest version.

`Error downloading http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.lang.Exception: Incomplete download
	at cc.arduino.utils.network.FileDownloader.readStreamCopyTo(FileDownloader.java:283)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:247)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
`

SpenceKonde, 2021-01-22:
megaTinyCore requires the latest version of the arduino avr-gcc toolchain package. In that message, the download error is occurring while board manager is trying to download http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip - that's a file stored on the official Arduino 

The file is available now as far as I can tell (and downloaded quickly); it may have been a transient issue, or it could be a problem with general internet access. 

If the error is persisting in the absence of general internet connectivity issues - could it be getting hung up on partial file in C:\Users\(your name)\AppData\Local\Arduino15\staging\packages (note that AppData is hidden) 
(You've verified thart you're not bumping into disk space limit or something, right?)

sideswiper22, 2021-01-23:
I have already reinstalled Arduino and deleted the Arduino15 folder prior to installation to clear any issues before I posted this issue. Yep, I've verified I still have a hundred plus gigs of storage space. Might just be a server error on Arduino's end. Will have another go.

closed 2021-01-23 by sideswiper22