SpenceKonde/megaTinyCore #303: programming problem with mEDBG and v2.2.6 ?

ffred, 2021-01-25:
Hi,
didn't use megaTinyCore recently and was still with v2.1.3.
I'm using an mEDBG (ATmega32U4) from an ATTINY416-XNANO borad for programming (3216 & 414 MCUs).

if I update to v2.2.6, I can't program anymore and got this error :
```
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
An error occurred while uploading the sketch
```

if I get back to v2.1.3, it's working again with this reply :
```
avrdude: jtag3_edbg_signoff(): unexpected response 0x03, 0x80
```

not a real problem actually for me. I can stay with v2.1.3 for what I'm doing..
was just to let you know.

thanks for your work.

SpenceKonde, 2021-01-25:
It could be either from the AVRdude upgrade (we moved to arduino18 from arduino16 to pick up support for PICkit and MPLab SNAP as programmers. First I've heard of it, but it is fairly new. I'd suggest asking on forums - that *was* the protocol those changes were made to, and so it's plausible that they could have broken things for older mEDBG devices in the process...

The only other change that the programmer would see is that some fuses are set with all UPDI uploads - but I think that the AVRdude upgrade is more likely the culprit. I don't use the official boards/programmers (I was 100% jtag2updi until pyupdi style serial port programming became an option during 2.2.0 development), ; I don't think I have any hardware I could see if it reproduces on - but all that could do is confirm the cause of the problem; I certainly couldn't solve it. (and AVRdude isn't being maintained anymore - things like this happen when someone *really* wants something changed, and I 

The 2.2.x versions have some pretty significant changes - ones which you probably want >.> 

 I think removing 2.2.x via board manager and doing manual installation of the core (I think that's described in https://github.com/SpenceKonde/megaTinyCore/blob/master/Installation.md ) would work around it if it's an AVRdude problem.

ffred, 2021-01-26:
ok, thanks.. will see if I can do some more tests, but eventually will probably use an MPLAB SNAP.

ffred, 2021-02-02:
just wanted to try with programming with an MPLAB SNAP after updating to v2.2.6, but it doesn't seem to be knwow.
did I miss something ??

```
An error occurred while uploading the sketch

avrdude: Can't find programmer id "snap_updi"

Valid programmers are:
  2232HIO          = FT2232H based generic programmer
  4232h            = FT4232H based generic programmer
  89isp            = Atmel at89isp cable
  abcmini          = ABCmini Board, aka Dick Smith HOTCHIP
  alf              = Nightshade ALF-PgmAVR, http://nightshade.homeip.net/
  arduino          = Arduino
  arduino-ft232r   = Arduino: FT232R connected to ISP
  atisp            = AT-ISP V1.1 programming cable for AVR-SDK1 from <http://micro-research.co.th/> 
  atmelice         = Atmel-ICE (ARM/AVR) in JTAG mode
  atmelice_dw      = Atmel-ICE (ARM/AVR) in debugWIRE mode
  atmelice_isp     = Atmel-ICE (ARM/AVR) in ISP mode
  atmelice_pdi     = Atmel-ICE (ARM/AVR) in PDI mode
  atmelice_updi    = Atmel-ICE (ARM/AVR) in UPDI mode
  avr109           = Atmel AppNote AVR109 Boot Loader
....
```

ffred, 2021-02-02:
just search into files installed and found an avrdude.conf containing `id  = "snap_updi"; ` so I probably have mixed files from the different updates..
will try to clean all that. maybe uninstall most and do a clean install... will see.

closed 2021-02-02 by ffred